Xilinx/CHaiDNN #2: Bitstream Integration error building CHaiDNN Hardware from source

jlamperez, 2018-03-25:
Hi, 

I am trying to build CHaiDNN HW from source and when it starts generating the bitstream I have this error

```
[12:43:53] Starting bitstream generation..

ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, write_bitstream ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/jlamperez/Vivado/Vivado2017.4/SDx/SDx/2017.4/bin/vpl  --iprepo /home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/design/scripts/kernel/_sds/iprepo/repo  --iprepo /home/jlamperez/Vivado/Vivado2017.4/SDx/SDx/2017.4/data/ip/xilinx  --platform /home/jlamperez/Vivado/Vivado2017.4/SDx/SDx/2017.4/platforms/zcu102/zcu102.xpfm  --temp_dir /home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/design/scripts/kernel/_sds/p0  --output_dir /home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/design/scripts/kernel/_sds/p0/vpl  --input_file /home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/design/scripts/kernel/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels XiConvolutionTop:PoolTop:xiSgemvTop:XiDeconvTop --webtalk_flag SDSoC  --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" '
sds++ log file saved as /home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/design/scripts/kernel/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:77: fallo en las instrucciones para el objetivo 'libxlnxdnn.so'
make: *** [libxlnxdnn.so] Error 1

```

Any way to solve it?

Thanks in advance,

Jorge

jlamperez, 2018-03-25:
I have built the project again from the beginning and it has worked.

Closing this issue.

closed 2018-03-25 by jlamperez

HeroGian, 2018-03-30:
Hi, I am trying to build CHaiDNN from source with the Makefile provided, but I have the same problem that you have reported:

`ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, write_bitstream ERROR
`

Could you tell me how you solved the problem? I have tried to rebuild but the problem persists.

Thank you in advance.

jlamperez, 2018-03-30:
The only thing different I did regarding to the first build was to remove the #  [here](https://github.com/Xilinx/CHaiDNN/blob/master/design/scripts/kernel/Makefile#L41) in lines 41 and 48.

Then I rebuilt and it worked

Radhikamkr, 2019-01-05:
Hi,
I am getting the same error.
Can you please tell me how to solve it?
The link that you provided is not accessible to me.

Thanks

JIAWENYANG, 2019-01-14:
Sorry, but it appears that a Xilinx program has terminated unexpectedly. Please contact Xilinx technical support for further assistance and give them the contents of @jlamperez 
[chaindnn_Release.build.log](https://github.com/Xilinx/CHaiDNN/files/2754225/chaindnn_Release.build.log)