Xilinx/CHaiDNN #3: 'clock_start': symbol multiply defined error building CHaiDNN using SDx IDE

jlamperez, 2018-03-25:
Hi,

I am trying to build CHaiDNN with SDx but I have the next issue

```
Building target: CHaiDNN_new.elf
Invoking: SDS++ Linker
sds++ -xp param:compiler.skipTimingCheckAndFrequencyScaling=1 -L/home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/SD_Card/opencv_arm64/lib -L/home/jlamperez/Proyectos/CHaiDNN/CHaiDNN/SD_Card/protobuf_arm64/lib --remote_ip_cache /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/ip_cache -o "CHaiDNN_new.elf"  ./src/software/xtract/caffe.pb.o ./src/software/xtract/caffe_network_parser.o ./src/software/xtract/hw_settings.o ./src/software/xtract/xgraph.o ./src/software/xtract/xi_funcs.o ./src/software/xtract/xtract_opt.o  ./src/software/swkernels/xi_crop.o ./src/software/swkernels/xi_nms_top.o ./src/software/swkernels/xi_norm_top.o ./src/software/swkernels/xi_permute_top.o ./src/software/swkernels/xi_swsoftmax_top.o  ./src/software/scheduler/xi_scheduler.o ./src/software/scheduler/xi_thread_routines.o ./src/software/scheduler/xi_utils.o  ./src/software/interface/xi_interface.o  ./src/software/init/xi_init.o  ./src/software/include/xchange_structs.o  ./src/software/imageread/xi_input_image.o  ./src/software/example/advanced_resnet50_ex.o ./src/software/example/alexnet_ex.o ./src/software/example/alexnetfcn_ex.o ./src/software/example/googlenet_ex.o ./src/software/example/ssd_ex.o ./src/software/example/vgg_ex.o  ./src/software/common/xi_kernels.o  ./src/software/checkers/checkers.o  ./src/software/bufmgmt/xi_buf_mgmt.o  ./src/design/utils/common/dnn_wrapper.o  ./src/design/src/pool/example/pooling_layer_dp_2xio_top.o  ./src/design/src/gemv/example/xi_sgemv_top.o  ./src/design/src/deconv/example/xi_deconv_top.o  ./src/design/src/conv/example/xi_convolution_top.o   -lopencv_core -llzma -ltiff -lpng16 -lz -ljpeg -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -lwebp -lprotobuf -dmclkid 3  -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf "zcu102"
Removing implementation files from previous run
Analyzing object files
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/caffe.pb.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/caffe_network_parser.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/hw_settings.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/xgraph.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/xi_funcs.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/xtract/xtract_opt.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/swkernels/xi_crop.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/swkernels/xi_nms_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/swkernels/xi_norm_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/swkernels/xi_permute_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/swkernels/xi_swsoftmax_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/scheduler/xi_scheduler.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/scheduler/xi_thread_routines.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/scheduler/xi_utils.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/interface/xi_interface.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/init/xi_init.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/include/xchange_structs.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/imageread/xi_input_image.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/advanced_resnet50_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/alexnet_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/alexnetfcn_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/googlenet_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/ssd_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/example/vgg_ex.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/common/xi_kernels.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/checkers/checkers.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/software/bufmgmt/xi_buf_mgmt.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/design/utils/common/dnn_wrapper.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/design/src/pool/example/pooling_layer_dp_2xio_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/design/src/gemv/example/xi_sgemv_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/design/src/deconv/example/xi_deconv_top.o
... /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/src/design/src/conv/example/xi_convolution_top.o
Generating data motion network
ERROR: Linking globals named 'clock_start': symbol multiply defined!
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/jlamperez/Vivado/Vivado2017.4/SDx/SDx/2017.4/bin/llvm-link -o /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/sds_all.o /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/caffe.pb.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/caffe_network_parser.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/hw_settings.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/xgraph.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/xi_funcs.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/xtract/xtract_opt.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/swkernels/xi_crop.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/swkernels/xi_nms_top.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/swkernels/xi_norm_top.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/swkernels/xi_permute_top.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/swkernels/xi_swsoftmax_top.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/scheduler/xi_scheduler.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/scheduler/xi_thread_routines.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/scheduler/xi_utils.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/interface/xi_interface.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/init/xi_init.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/include/xchange_structs.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/imageread/xi_input_image.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/advanced_resnet50_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/alexnet_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/alexnetfcn_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/googlenet_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/ssd_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/example/vgg_ex.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/common/xi_kernels.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/checkers/checkers.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/software/bufmgmt/xi_buf_mgmt.s /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/.llvm/./src/design/utils/common/dnn_wrapper.s /home/j [... cut ...] ds/.llvm/./src/design/src/conv/example/xi_convolution_top.s'
sds++ log file saved as /home/jlamperez/Proyectos/CHaiDNN/SDx_ws/CHaiDNN_new/Debug/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: fallo en las instrucciones para el objetivo 'CHaiDNN_new.elf'
make: *** [CHaiDNN_new.elf] Error 1
```

It seems that clock_start  symbol is multiply defined. Do I need to add something else to the linker appart  from what is written in the build CHaiDNN with SDx?

Thanks,

Jorge

jlamperez, 2018-03-25:
I have seen that this multiple definitions are located in this files in `CHaiDNN_new/src/software/example/`:

* alexnet_ex.cpp
* advanced_resnet50_ex.cpp
* alexnetfcn_ex.cpp
* googlenet_ex.cpp
* ssd_ex.cpp
* vgg_ex.cpp

Removing all but one, for instance alexnet_ex.cpp, using exclude from build creates the sd_card folder with the corresponding files but instead of having the `libxlnxdnn.so` I have a CHaiDNN.elf executable.

Is this procedure correct? 

How can I build `libxlnxdnn.so` using SDx?

Thanks,

Jorge

PhoenixShield, 2018-04-12:
I meet the same error while building the project. Could anyone help us on this? Thanks!

VishalX, 2018-04-18:
[Please check here](https://github.com/Xilinx/CHaiDNN/issues/9)

commit e78c1457 (2018-07-26): Update README with correct paths and notes.

Issue close #9
Issue close #3
closed 2018-04-18 by VishalX