Xilinx/CHaiDNN #4: Integration error, problem implementing dynamic region

OnTimeSir, 2018-04-03:
Hi,

I am trying to build CHaiDNN with SDx but I have the issue below：

ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, opt_design ERROR
make: *** [test_alexnet.elf] Error 1
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/zhoufanyu/Programes/Xilinx/SDx/2017.4/bin/vpl  --iprepo /home/zhoufanyu/workspace/test_alexnet/Debug/_sds/iprepo/repo  --iprepo /home/zhoufanyu/Programes/Xilinx/SDx/2017.4/data/ip/xilinx  --platform /home/zhoufanyu/Programes/Xilinx/SDx/2017.4/platforms/zcu102/zcu102.xpfm  --temp_dir /home/zhoufanyu/workspace/test_alexnet/Debug/_sds/p0  --output_dir /home/zhoufanyu/workspace/test_alexnet/Debug/_sds/p0/vpl  --input_file /home/zhoufanyu/workspace/test_alexnet/Debug/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:xiSgemvTop:XiDeconvTop:XiConvolutionTop --webtalk_flag SDSoC  --remote_ip_cache /home/zhoufanyu/workspace/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" '
sds++ log file saved as /home/zhoufanyu/workspace/test_alexnet/Debug/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: recipe for target 'test_alexnet.elf' failed

Could anyone help? Thanks.

nkpavan, 2018-04-20:
Hi

Please re-build the design and if you still face the same issue please send the vivado.log file.

closed 2018-07-25 by VishalX

tunghoang290780, 2019-12-20:
Has anyone solved this problem "ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, write_bitstream ERROR". I did compile ChainDNN with Ubuntu14.10, SDSoC 2018.2 and ZCU102 board but after sync with upstream repo, this error apparently appeared.

xindubawukong, 2020-02-20:
I have this problem too. I think it may because there's no enough memory.

tunghoang290780, 2020-02-20:
> I have this problem too. I think it may because there's no enough memory.

Hi @xindubawukong: Do you mean that zc102 board does not have enough SRAM memory?

xindubawukong, 2020-02-21:
@tunghoang290780 , I mean there's no enough RAM memory on your Ubuntu. This is a possible reason. SDx needs a lot of memory to run. If you are on a 8GB or less memory machine, you can try 16GB or larger.

tunghoang290780, 2020-02-21:
> @tunghoang290780 , I mean there's no enough RAM memory on your Ubuntu. This is a possible reason. SDx needs a lot of memory to run. If you are on a 8GB or less memory machine, you can try 16GB or larger.

@xindubawukong that is possible due to another reason. I tried with two machines which both have >64GB DRAM.

yarenty, 2020-03-05:
Have the same issue with another library - xgboots, if you find any solution please share!

tunghoang290780, 2020-03-05:
@yarenty 
I fixed this
1) Modify xi_conv_config.h
#define XI_BIAS_URAM_EN          0      
#define XI_WTS_URAM_EN          1
#define XI_ISTG_URAM_EN          1
#define XI_OSTG_URAM_EN        1

2) Run command "
make POOL_ENABLE=0 DECONV_ENABLE=0 -j<N> 
where N is the number of cores

Okaymaddy, 2020-06-03:
@tunghoang290780 What N?

tunghoang290780, 2020-06-03:
@Okaymaddy 
The full command is `make POOL_ENABLE=0 DECONV_ENABLE=0 -jN` where N is the number of cores of your machine

Okaymaddy, 2020-06-03:
@tunghoang290780 I am receiving some errors when I am trying to build ChaiDNN. I followed your instructions and made the changes in xi_conv_config.h. I am running for zcu104 in SDX 2019.1.
Attached is my error: 
[image]

Any idea how to fix this?

tunghoang290780, 2020-06-03:
@Okaymaddy I've got failures with SDX > 2019. Last version I was successful is SDSoC 2018.2

Okaymaddy, 2020-06-03:
@tunghoang290780 I see, I will try it in version 2018.2. Did you try to build for zcu104? Did all the steps mentioned here (https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md)  plus a few changes in xi_conv_top.h was enough to build the project successfully?

Okaymaddy, 2020-06-05:
@tunghoang290780 Can you tell me what was your machine's RAM and Memory when you were able build ChaiDNN?