Xilinx/CHaiDNN #6: Ristretto for ssd?

HelloSongRen, 2018-04-03:
How do you use ristretto to determine the precision parameters for ssd? Can ristretto fine-tune detection models like ssd? Thinks...

smenon009, 2018-04-20:
Hello,
By Default, Caffe-Ristretto version do not support SSD layers. We will need to add the SSD layers implementations manually to the Ristretto -Caffe and make sure the merged Caffe-Risterrto+SSD code-base does not complain and throw errors while building/ running.  Then, one could use  it to fine-tune/ re-train SSD type networks. CHaiDNN does not expect quantization parameters for SSD specific layers such as permute, normalize, flatten, NMS etc.

HelloSongRen, 2018-04-27:
Hello, can you share your merged Caffe-Risterrto+SSD code?

smenon009, 2018-06-04:
Hi @HelloSongRen 
Will check and get back on this.

smenon009, 2018-07-25:
Hi @HelloSongRen 
We will soon releasing some tools for re-training as well and that will support SSD.

closed 2018-07-25 by smenon009