Xilinx/CHaiDNN #8: Re-building designs results in timing error

hpollittsmith, 2018-04-11:
Hi,

When I rebuild the reference designs, either from the scripts or in SDSOC 2017.4, the resulting implementation has timing violations. I find I need to set the speed of the hardware functions to 100MHz rather than the 200MHz in the instructions. Particularly for the scripts method, I do nothing more than run the instructions provided in the README file. Is this to be expected, or am I missing some extra step?

hpollittsmith, 2018-04-19:
Hi again,

I modified the Makefile (CHaiDNN-master/design/scripts/kernel/Makefile) to change the implementation strategy from default to Performance_Explore and this time it meets timing. Specifically, I modified the following line:

SDSFLAGS = -sds-pf ${PLATFORM} _-impl-strategy Performance_Explore_ -xp param:compiler.skipTimingCheckAndFrequencyScaling=1 \

When building the design in the SDx GUI, in the project C/C++ Build Settings, in the **SDS++ Linker Command field** add **_-impl-strategy Performance Explore_**, clean and build the project and it works.

nkpavan, 2018-04-20:
Are you adding any other IPs other than the ones provided?

hpollittsmith, 2018-04-20:
Hi, no, I was just implementing the examples as provided in Github.

nkpavan, 2018-04-20:
Please send the timing report.

hpollittsmith, 2018-04-20:
Hi,

Here is it.
[updated_full_design_timing_summary_routed.rpt.txt](https://github.com/Xilinx/CHaiDNN/files/1932740/updated_full_design_timing_summary_routed.rpt.txt)

closed 2018-07-25 by VishalX