Xilinx/CHaiDNN #9: 'main': symbol multiply defined error building CHaiDNN using SDx IDE

ensafi, 2018-04-12:
Following the [BUILD_USING_SDX_GUI instructions](https://github.com/Xilinx/CHaiDNN/blob/master/BUILD_USING_SDX_GUI.md) and working around issue #3 by making `clock_start` and related variables `static`, I am getting a final link error with `main()` being multiply defined:

ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2017.4/bin/llvm-link -o /workspace/CHaiDNN/Debug/_sds/.llvm/sds_all.o /workspace/CHaiDNN/Debug/_sds/.llvm/./src/software/xtract/caffe.pb.s ...

hpollittsmith, 2018-04-12:
I've found that you cannot build all of the applications in SDx at the same time. If you want to build the SSD example, then you should exclude the other applications from the build. For example, in the Project Explorer-->src-->software-->example right-click on alexnet.cpp and in the menu, select Resource Configurations-->Exclude from Build...   Check the "Debug" and "Release" boxes. The alexnet_ex.cpp will be greyed out in the Project Explorer window. Do this for all the example applications (except the one you want to compile).

Note that this will create a elf file with the same name as your project.

VishalX, 2018-04-18:
Please exclude other example files from the project; Use one example file while building.

commit e78c1457 (2018-07-26): Update README with correct paths and notes.

Issue close #9
Issue close #3
closed 2018-04-18 by VishalX