Xilinx/CHaiDNN #10: New network inference doesn't work

NeutrinoXY, 2018-04-19:
Hello,

When I try to run my own network, a simple LeNet trained on MNIST, all outputs are barely equal to 0.1 (10 ouputs).

I used the loadImagetoBuffptr function in the main function, with a mean.txt file (0 for each channel because I didn't substract the mean for my training). My input is a 28x28 .png image from the MNIST database, in the RGB (opened as BGR with opencv) format (so the 3 channels are the same).

I tried so much things, but I'm still not able to make it work. The network runs well when using python and caffe-ristretto, and all provided .elf examples work.

pptnz, 2018-07-18:
Hello,
I also ran my own network and I have the similar problem that all outputs are equal to 0.5 for 2 outputs.
If you have solved the problem, can you let me know what you did?

Thank you.

NeutrinoXY, 2018-07-18:
Hello,

I just fixed the code. But there are still bugs I didn't fix yet. I also need to clean my code before I can send a pull request. Probably the next month when I get some time for that.

pptnz, 2018-07-20:
Thank you for your comment. I am looking forward to having your code.
Before the pull request, I will try to fix the code, so could you tell me which part of a code you fixed?

shinya321, 2019-02-14:
Hi, I have the same problem that all outputs are 0.5 for 2 outputs.
If you could already solve it, could u plz explain how to do ?
deploy.pt is [here](https://github.com/shinya321/caffe-prototxt-for-CHaiDNN/blob/master/fcn5_deploy.prototxt)

Best regards