Xilinx/CHaiDNN #12: Problem while building src  in SDx

lishen565, 2018-05-02:


lishen565, 2018-05-06:
Hi, I've build the projects in ubuntu 17.10 but it still failed when starting bitstream generation . The console output is below:
[20:02:36] Synthesis is running.
[20:03:36] Synthesis is running.
[20:04:36] Synthesis is running.
[20:05:36] Synthesis is running.
[20:06:36] Synthesis is running.
[20:07:36] Synthesis is running.
[20:08:36] Synthesis is running.
[20:09:36] Synthesis is running.
[20:10:36] Synthesis is running.
[20:17:36] Finished 1st of 6 tasks (FPGA synthesis). Elapsed time: 05h 04m 22s 

[20:17:36] Starting to link synthesized kernels..
[20:19:31] Phase 1 Generating reports.
[20:19:31] Phase 2 Done generating reports.
[20:31:06] Finished 2nd of 6 tasks (FPGA linking synthesized kernels to platform). Elapsed time: 00h 13m 29s 

[20:31:06] Starting logic optimization..
[20:32:29] Phase 1 Retarget
[20:32:47] Phase 2 Constant propagation
[20:32:59] Phase 3 Sweep
[20:33:17] Phase 4 BUFG optimization
[20:33:34] Phase 5 Shift Register Optimization
[00:36:49] Finished 3rd of 6 tasks (FPGA logic optimization). Elapsed time: 04h 05m 43s 

[00:36:49] Starting logic placement..
[00:37:02] Phase 1 Placer Initialization
[00:37:02] Phase 1.1 Placer Initialization Netlist Sorting
[00:40:01] Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
[00:42:11] Phase 1.3 Build Placer Netlist Model
[02:11:33] Phase 1.4 Constrain Clocks/Macros
[02:12:55] Phase 2 Global Placement
[02:33:43] Phase 3 Detail Placement
[02:33:43] Phase 3.1 Commit Multi Column Macros
[02:34:34] Phase 3.2 Commit Most Macros & LUTRAMs
[02:35:56] Phase 3.3 Area Swap Optimization
[02:36:03] Phase 3.4 Pipeline Register Optimization
[02:36:09] Phase 3.5 IO Cut Optimizer
[02:36:15] Phase 3.6 Timing Path Optimizer
[02:36:15] Phase 3.7 Fast Optimization
[02:36:54] Phase 3.8 Small Shape Clustering
[02:37:39] Phase 3.9 DP Optimization
[02:39:04] Phase 3.10 Flow Legalize Slice Clusters
[02:39:04] Phase 3.11 Slice Area Swap
[02:39:36] Phase 3.12 Commit Slice Clusters
[02:40:08] Phase 3.13 Re-assign LUT pins
[02:40:21] Phase 3.14 Pipeline Register Optimization
[02:40:27] Phase 3.15 Fast Optimization
[02:42:02] Phase 4 Post Placement Optimization and Clean-Up
[02:42:02] Phase 4.1 Post Commit Optimization
[02:47:42] Phase 4.1.1 Post Placement Optimization
[02:47:54] Phase 4.1.1.1 BUFG Insertion
[02:51:30] Phase 4.2 Post Placement Cleanup
[02:51:48] Phase 4.3 Placer Reporting
[02:51:54] Phase 4.4 Final Placement Cleanup
[02:52:27] Finished 4th of 6 tasks (FPGA logic placement). Elapsed time: 02h 15m 37s 

[02:52:27] Starting logic routing..
[02:53:23] Phase 1 Build RT Design
[02:54:42] Phase 2 Router Initialization
[02:54:42] Phase 2.1 Fix Topology Constraints
[02:54:48] Phase 2.2 Pre Route Cleanup
[02:54:48] Phase 2.3 Global Clock Net Routing
[02:55:07] Phase 2.4 Update Timing
[03:09:20] Phase 3 Initial Routing
[03:09:20] Phase 3.1 Global Routing
[03:18:34] Phase 4 Rip-up And Reroute
[03:18:34] Phase 4.1 Global Iteration 0
[04:01:53] Phase 4.2 Global Iteration 1
[04:08:25] Phase 4.3 Global Iteration 2
[04:12:08] Phase 4.4 Global Iteration 3
[04:13:44] Phase 5 Delay and Skew Optimization
[04:13:44] Phase 5.1 Delay CleanUp
[04:13:44] Phase 5.1.1 Update Timing
[04:17:53] Phase 5.2 Clock Skew Optimization
[04:17:53] Phase 6 Post Hold Fix
[04:17:53] Phase 6.1 Hold Fix Iter
[04:18:00] Phase 6.1.1 Update Timing
[04:18:51] Phase 7 Route finalize
[04:18:58] Phase 8 Verifying routed nets
[04:18:58] Phase 9 Depositing Routes
[04:20:07] Phase 10 Post Router Timing
[04:21:11] Finished 5th of 6 tasks (FPGA routing). Elapsed time: 01h 28m 43s 

[04:21:11] Starting bitstream generation..

_ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, route_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2017.4/bin/vpl  --iprepo /home/lishen/workspace_sdx/CHaiDNN/Debug/_sds/iprepo/repo  --iprepo /opt/Xilinx/SDx/2017.4/data/ip/xilinx  --platform /opt/Xilinx/SDx/2017.4/platforms/zcu102/zcu102.xpfm  --temp_dir /home/lishen/workspace_sdx/CHaiDNN/Debug/_sds/p0  --output_dir /home/lishen/workspace_sdx/CHaiDNN/Debug/_sds/p0/vpl  --input_file /home/lishen/workspace_sdx/CHaiDNN/Debug/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:xiSgemvTop:XiDeconvTop:XiConvolutionTop --webtalk_flag SDSoC  --remote_ip_cache /home/lishen/workspace_sdx/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" '
make: *** [CHaiDNN.elf] Error 1
sds++ log file saved as /home/lishen/workspace_sdx/CHaiDNN/Debug/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: recipe for target 'CHaiDNN.elf' failed_

lishen565, 2018-05-06:
And there are many errors like 
"Multiple markers at this line 	- unused label 'Plane_1x1_Loop' [-Wunused-label]", but SDx IDE building procedure didn't stop and continued to Creating Vivado project and starting FPGA synthesis. Is this right? Thanks a lot.

HelloSongRen, 2018-05-29:
I got the same problem, have you solved it?@lishen565

VishalX, 2018-07-25:
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, route_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker

Issue with SDx 2017.4. Fixed with SDx 18.2.

closed 2018-07-25 by VishalX