Xilinx/CHaiDNN #14: Questions about the size of the .elf files and build time.

lishen565, 2018-05-16:
Hi all, I've test the example elf in our project and they work well. Then I had builded the project in SDx GUI and found that the elf was about 10MB. But the example elf in CHaiDNN project is about 59kB. Why my elf file which produced by SDx is much larger than yours?
The six example elf you provided could work well on the same FPGA bitstream. So I think the example elf  has few relationship about the bitstream. I don't want to rebuild the whole project just as I  modified the application code, because it would take a very long time to complete the build procedure. Do you have any good way to shorten the rebuild time?
Thanks a lot!

NeutrinoXY, 2018-05-22:
Hello,

When you build your project, you can choose to create a library file instead of creating a single elf for your application. Thus, you can compile your application separately, and it takes a few seconds. This is how the provided .elf examples have been built.

You can choose this option in SDx GUI, or you can use the provided makefile.

Best regards.

lishen565, 2018-05-24:
Thanks a lot. I'll try that as your advice.

VishalX, 2018-07-25:
Closing this issue.

closed 2018-07-25 by VishalX