Xilinx/CHaiDNN #15: Does CHaiDNN support Faster R-CNN?

Kang-An, 2018-05-18:
Hi, I define a faster r-cnn model followed by RUN_NEW_NETWORK.md, but it can't run correctly.

`[libprotobuf ERROR google/protobuf/text_format.cc:299] Error parsing text-format caffe.NetParameter: 3722:25: Message type "caffe.LayerParameter" has no field named "psroi_pooling_param".
[EP002] Couldn't Parse the deploy file : /mnt/models/CustomNet/deploy.prototxt`

So I want to know if CHAiDNN supports roi layer?

VishalX, 2018-05-29:
Hi @Kang-An,

Current CHaiDNN release doesn't support ROI layer.

Kang-An, 2018-05-29:
yes, roi pooling and PSROIPooling is not declared in `caffe_layer_map.hpp`.
so i'm trying to add these layer in CHaiDNN, can you give me some suggestions?

I have figured out the overall flow, but I can't understand the `poolGeneral ` function in pool_dp_2xio. Why it need two input buffers and two output buffers?

By the way, please forgive my lame English...

Kang-An, 2018-05-29:
yes, roi pooling and PSROIPooling is not declared in caffe_layer_map.hpp.
so i'm trying to add these layer in CHaiDNN, can you give me some   suggestions?
@VishalX 
I have figured out the overall flow, but I can't understand the poolGeneral function in pool_dp_2xio. Why it need two input buffers and two output buffers?

By the way, please forgive my lame English...

Kang-An, 2018-05-29:
yes, roi pooling and PSROIPooling is not declared in `caffe_layer_map.hpp`.
so i'm trying to add these layer in CHaiDNN, can you give me some suggestions?

I have figured out the overall flow, but I can't understand the `poolGeneral `function in pool_dp_2xio. Why it need two input buffers and two output buffers?
@VishalX 

By the way, please forgive my lame English...

VishalX, 2018-05-31:
The implementation of pool depends on how the input/output data is organized. For Pooling, i/o data is organized in two DDR buffers. For higher performance, it uses two different i/o buffers so that they can be used at the same time. 

It can be done using single buffer too. But, the performance will be lesser.

smenon009, 2018-07-25:
Hi @Kang-An 
Checkout the CHai v2 with enhanced features. Software layer plugin feature enables users to add custom layer implementations as software layers. Documentations available here:  https://github.com/Xilinx/CHaiDNN/blob/master/docs/SOFTWARE_LAYER_PLUGIN.md

closed 2018-07-25 by smenon009

Kang-An, 2018-07-26:
HI, @smenon009 
Hi, I've read the steps to create a custom software layer plugin, which says "custom layer can take multiple bottom blobs, but only one top blob", The example code accesses a bottom result by In_ptr = (float*)xchange->in_ptrs[0],

I want to apply Feature Pyramid Network for generate object proposal, it has eight bottoms for proposal layer. So i want to konw how to access multiple bottom results?
By the way, can you introduce the differences betweenin_ptr[0], in_ptr[1], in_ptr[2], in_ptr[3], and in_ptr[4]

thank for your reply.