Xilinx/CHaiDNN #16: makefile issue

Dominatrix007, 2018-05-29:
Hi I am trying to build the LeNet for reVISION platform. I wrote the cpp code and the Makefile as was given in the documentation. When I tried running the makefile I am getting this error:

/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find ../../lib/libc.so.6
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find ../../lib/ld-linux-aarch64.so.1
collect2: error: ld returned 1 exit status
Makefile:17: recipe for target 'LeNet.elf' failed
make: *** [LeNet.elf] Error 1

Although the necessary files are located in the given directory it was not able to locate it.
It would be really nice if you can help me resolve this issue.

Thanks,
Anjali

VishalX, 2018-05-29:
Hi Anjali,

Could you share the makefile?

Dominatrix007, 2018-05-29:
@VishalX  
Hi vishal,
Yes sure. I am attaching the Makefile. 
[Makefile.txt](https://github.com/Xilinx/CHaiDNN/files/2047897/Makefile.txt)

Thanks!

closed 2018-05-30 by Anjali-iit8