Xilinx/CHaiDNN #17: New project

Anjali-iit8, 2018-05-31:
Hi,
I have a project involving image processing and neural networks which I want to port to Xilinx platform zcu104_rv_ss. It was made using LeNet and I have successfully got the executable file for the same. Now I have a main code for which I want to get the elf file. Can you help me with that as I am unable to find any documentation regarding that.

Thanks,
Anjali

NeutrinoXY, 2018-05-31:
Hello,

I don't understand your issue. Can you be more specific ? What is your "main code" you want to get the .elf file from ?

Best regards.

Anjali-iit8, 2018-06-01:
Hello,
I have a project implemented using OpenCV and caffe packages. It is a classification project. I have trained the model using LeNet. Now I want to port it to the FPGA for which I want to convert it so that I can transfer it to the SD-Card for bootup.

Thanks,

Anjali-iit8, 2018-06-01:
@NeutrinoXY  Hello, can I also execute these examples given in the repository as well as my my project using the zcu104 platform?

Thanks

VishalX, 2018-07-25:
CHaiDNN-v2 supports zcu104 platform. Please try the latest release.

closed 2018-07-25 by VishalX