Xilinx/CHaiDNN #18: run a new caffe module ,illegal memory access

liyang53719, 2018-05-31:
when i try a new caffe module .i am done with SDSoC  .i put the file in the SD card and run.
 there is an ERROR,which is write down below.

ERROR: application performed illegal memory access and is being terminated

and there is a warning :

[libprotobuf INFO google/protobuf/io/coded_stream.cc:610] Reading dangerously large protocol message.  If the message turns out to be larger than 2147483647 bytes, parsing will be halted for security reasons.  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
[libprotobuf WARNING google/protobuf/io/coded_stream.cc:81] The total number of bytes read was 209274056

i do not how to fix it.please help me.

liyang53719, 2018-05-31:
I built the project in SDSoC using the methods provided on github. Reserved vgg_ex.c. Compile the project using HW acceleration. The generated file can be used. Then on this basis, I changed the path of the caffe model to my own. The malloc function is modified to the same size. HW compilation is not used. However, an error occurred.

NeutrinoXY, 2018-05-31:
Hi,

Can you provide your .cpp file calling the xiExec function ?

Best regards.

VishalX, 2018-06-01:
Hi @liyang53719,

Could you provide the log as to where this memory fault occurred? Parsing? Initialization? Execution?

liyang53719, 2018-06-01:
HI @VishalX 
This is the log."..." means almost same lines.

root@xilinx-zcu102-2017_4:/mnt# ./CHaiDNN_test.elf

 Create memory Done
[PARSE] Parsing conv1_1
...
...
...
[PARSE] Parsing conv7
[libprotobuf INFO google/protobuf/io/coded_stream.cc:610] Reading dangerously large protocol message.  If the message turns out to be larger than 2147483647 bytes, parsing will be halted for security reasons.  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
[libprotobuf WARNING google/protobuf/io/coded_stream.cc:81] The total number of bytes read was 209274056
[IG001] Saving /mnt/models/test/data/conv1_weights (4.43 MB)
....
....
....
[IG001] Saving /mnt/models/test/data/conv5_bias (76 B)
[INFOx] Graph generated
[INFOx] Generating JobQueue
ERROR: application performed illegal memory access and is being terminated
root@xilinx-zcu102-2017_4:/mnt#

VishalX, 2018-06-01:
Hi @liyang53719,

It seems the initialization is giving the memory fault. 
The protobuf warning that you are seeing has nothing to do with this and it's a known issue with protobuf. 

I would suggest inserting some debug statements inside initialization module to find out where exactly the error has occurred.

liyang53719, 2018-06-04:
HI @VishalX
After some testing, I think there is something wrong with "xgraphopt.xi_opt_concat(xgraph)".
The Caffe network I tested contained a "concat" layer with three "bottoms" and two "tops." The prototxt file is described as follows:
. . .
. . .
. . .
Layer {
   Name: "concat_stage"
   Type: "Concat"
   Bottom: "conv5_5_1"
   Bottom: "conv5_5_2"
   Bottom: "conv4_4"
   Top: "concat_stage"
   Concat_param {
     Axis: 1
   }
Layer {
   Name: "conv6_1_1"
   Type: "ConvolutionRistretto"
   Bottom: "concat_stage"
   Top: "conv6_1_1"
   ...
   ...
   ...
   }
Layer {
   Name: "conv6_1_2"
   Type: "ConvolutionRistretto"
   Bottom: "concat_stage"
   Top: "conv6_1_2"
   ...
   ...
   ...
   }
. . .
. . .
. . .
And I found some comments in the "XGraphOpt::xi_opt_concat" function, which read "//TODO we need think and add for multiple top scenarios".
Does this mean that my "multiple top concat" layer caused the error?

rhysaeron, 2018-06-25:
I am having the same issue and any help would be much appreciated

VishalX, 2018-10-11:
CHaiDNN-v1 issue. Closing. Please migrate to v2.

closed 2018-10-11 by VishalX