Xilinx/CHaiDNN #19: Different platform

Anjali-iit8, 2018-06-01:
Hello,
Can I execute these examples using a different platform, i.e. zcu104.
I tried implementing it but it gives an error regarding the clock ID. This line in the Makefile is giving this error. "-sds-hw XiConvolutionTop ../../src/conv/example/xi_convolution_top.cpp -clkid 3 -hls-tcl ./config_core2.tcl -sds-end -dmclkid 3 "
Can you tell me what this line means and also how can I change it with respective to zcu104 platform.

Thanks

NeutrinoXY, 2018-06-01:
Hi,

It's possible the clock ID 3 (200 MHz on ZCU102) doesn't exist for your platform.

You can get the clock IDs values with this command :

sdscc –sds-pf-info
<platform name>

Anyway, how much time do you have for your project ? You'll have to do many modifications to the code before you can build this project on the zcu104 platform.

Best regards.

Anjali-iit8, 2018-06-01:
I have to complete my project within two weeks. If you can help me build up my project as I am not able to find any documentation regarding this.

Thanks

NeutrinoXY, 2018-06-01:
The deadline is tight.

Send me an email at neutrinoxy@gmail.com, it will be easier to help you.

myih, 2018-06-20:
Hi @Anjali-iit8 @NeutrinoXY 
I'm also trying to build this on ZCU104, any suggestion on how to modify the code?
Thank you!

VishalX, 2018-07-25:
Try CHaiDNN-v2 which supports zcu104 platform. Please go through the documentation on how to build CHaiDNN for different configurations.

Thanks

closed 2018-07-25 by VishalX