Xilinx/CHaiDNN #23: The probability that a picture can be classified into different categories is equal, but it is normal on PC.

daichong123, 2018-06-08:
Hi,Thank you for your source, I have successfully run on the board. Now I have a problem. I use VGG16 to fine-tune to classify four kinds of objects. The classification effect on PC is very good, but the result of running on board is that the probabilities of these four classification are the same, that is 0.25. I just changed the modelfile and meanfile to my own, changed the num_output and input picture, the other has not changed.My deploy.prototxt and out.txt files are below.Can you help me? Thank you!
[deploy.prototxt.txt](https://github.com/Xilinx/CHaiDNN/files/2082827/deploy.prototxt.txt)
[out.txt](https://github.com/Xilinx/CHaiDNN/files/2082828/out.txt)

NeutrinoXY, 2018-06-08:
Hello,

This library is great, but there are a lot of bugs, and I had to fix some of them in order to run my own network.

But your problem could have many sources. Can you share your main .cpp file (the one with the main and "load image" functions), your input image and your .caffemodel file?

daichong123, 2018-06-08:
Hi,
Thank you for your reply. I have not change the main.cpp provided by ChaiDNN. I just fine-tuned the VGG16 model to classify the retina OCT images.The following is my prototxt and the input image,due to more than 10M, my model file cannot be uploaded
[file.zip](https://github.com/Xilinx/CHaiDNN/files/2084039/file.zip)

NeutrinoXY, 2018-06-13:
Hi,

Please take a look at : https://github.com/Xilinx/CHaiDNN/issues/25#issuecomment-396529607

It could help you.

closed 2018-10-11 by VishalX