Xilinx/CHaiDNN #30: Different platform: ps_e_S_AXI_HP0_FPD for in1 cannot be found in the platform!

HelloSongRen, 2018-06-25:
Hi,
I am trying to import CHaiDNN to a xfopencv demo with HDMI input named filter2d, and build CHaiDNN to a shard library. But the platform of filter2d is zcu102_es2_rv_ss, not zcu102.
When I build CHaiDNN without  hardware accelerator, there is nothing wrong. When I accelerate the XiConvolutionTop, I get some errors:

WARNING: [DMAnalysis 83-4492] Unable to determine the memory attributes passed to scalar_conv_args of function XiConvolutionTop at /home/ssssspe/FPGA/workspace/filter2d/src/design/utils/common/dnn_wrapper.cpp:59, please use mem_attribute pragma to specify
CRITICAL WARNING: [DMAnalysis 83-4483] Cannot find corresponding HW caller for sds_wait(1); @ /home/ssssspe/FPGA/workspace/filter2d/src/software/scheduler/xi_scheduler.cpp:587, application may die at run time
INFO: [DMAnalysis 83-4444] Scheduling data transfer graph for partition 0
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for in1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP1_FPD for in2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP2_FPD for out1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP3_FPD for out2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP1_FPD for weights2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP2_FPD for output1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP3_FPD for output2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP2_FPD for input_other1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP3_FPD for input_other2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP2_FPD for input_1st cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP2_FPD for bias cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP3_FPD for inp_norm_2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP3_FPD for inp_norm_3 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for istg_out1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP1_FPD for istg_out2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP1_FPD for weights2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for input cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for bias cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for output cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: ps_e_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4445] Failed scheduling data transfer graph!
/home/ssssspe/FPGA/SDx/2017.4/bin/XidanePass: 1: /home/ssssspe/FPGA/SDx/2017.4/bin/XidanePass: gawk: not found
Data motion generation exited with return code 1
- errors detected
sds++ log file saved as /home/ssssspe/FPGA/workspace/filter2d/Debug/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: recipe for target 'libfilter2d.so' failed
make: *** [libfilter2d.so] Error 1

Do you have any suggestions? Thanks!

VishalX, 2018-06-25:
Hi @HelloSongRen,

HP port names are different in zcu102_rv_ss platform when compared to base platform. They need to be changed in the design files according to reVISION platform. 

Base platform HP port name:
ps_e_S_AXI_HP*_FPD

reVISION platform HP port name:
zynq_ultra_ps_e_0_S_AXI_HP*_FPD

where * is from 0 to 3. 

Please rename them accordingly and try to build it again. Also, please keep in mind that CHaiDNN uses about 90% BRAMs and some video IPs take up resources on PL and might affect PnR. Try to build them at a lower frequency say 150 MHz.

HelloSongRen, 2018-06-25:
Hi @VishalX ,
Thank you very much for your help!
It works, but there is still some errors. It seems that the reVISION platform do not include zynq_ultra_ps_e_0_S_AXI_HP0_FPD and zynq_ultra_ps_e_0_S_AXI_HP1_FPD?

ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for in1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP1_FPD for in2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP1_FPD for weights2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for istg_out1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP1_FPD for istg_out2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP1_FPD for weights2 cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for input cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for bias cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for output cannot be found in the platform!
ERROR: [DMAnalysis 83-4416] Specified sys_port: zynq_ultra_ps_e_0_S_AXI_HP0_FPD for weights1 cannot be found in the platform!
ERROR: [DMAnalysis 83-4445] Failed scheduling data transfer graph!

VishalX, 2018-06-25:
Hi @HelloSongRen,

In reVISION platform, *zynq_ultra_ps_e_0_S_AXI_HP0_FPD* & *zynq_ultra_ps_e_0_S_AXI_HP1_FPD* are reserved for video and IPs and they are not accessible. Could you please use 2 and 3 for all other ports too? This will build the design, but with only 2 HP ports, the performance of the design might degrade.

HelloSongRen, 2018-06-25:
Hi @VishalX,
Thank you very much for your help! I will try it.

HelloSongRen, 2018-06-26:
Hi @VishalX,

It is helpful, thank you!
And I get some errors when creating Vivado project and starting FPGA synthesis.

ERROR: [Common 17-179] Fork failed: Cannot allocate memory
ERROR: [Common 17-179] Fork failed: Cannot allocate memory

Here is my Vivado.log. 
[vivado.log](https://github.com/Xilinx/CHaiDNN/files/2135423/vivado.log)

I'm so sorry to trouble you, thank you!

VishalX, 2018-06-26:
@HelloSongRen ,

Seems like you are running out of memory. Please restart your machine and try again. Let me know if you still face this issue.

HelloSongRen, 2018-06-26:
Hi @VishalX,

I still face this issue after restarting my machine. I think it might be because that my computer only has 8G memory. How much memory is needed to build projects like this one?

VishalX, 2018-06-26:
@HelloSongRen,
Are you running any other builds along with CHaiDNN? If you have multiple SDx build instances, you might run out of memory. Try closing other applications when building CHaiDNN. 
Also, if you have a machine with larger RAM, try building on it.

HelloSongRen, 2018-06-27:
Hi @VishalX,

The errors about memory have been solved with your suggestions, thank you!

But I get a new error when it comes to the 3rd of 6 tasks:

......
[11:22:08] Starting logic optimization..
[11:23:53] Phase 1 Retarget
[11:24:29] Phase 2 Constant propagation
[11:24:45] Phase 3 Sweep
[11:25:16] Phase 4 BUFG optimization
[11:25:42] Phase 5 Shift Register Optimization

ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, opt_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/ssssspe/FPGA/SDx/2017.4/bin/vpl  --iprepo /home/ssssspe/FPGA/workspace/filter2d/Release/_sds/iprepo/repo  --iprepo /home/ssssspe/FPGA/SDx/2017.4/data/ip/xilinx  --platform /home/ssssspe/song/FPGA/zcu102_es2_rv_ss/zcu102_es2_rv_ss.xpfm  --temp_dir /home/ssssspe/FPGA/workspace/filter2d/Release/_sds/p0  --output_dir /home/ssssspe/FPGA/workspace/filter2d/Release/_sds/p0/vpl  --input_file /home/ssssspe/FPGA/workspace/filter2d/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:xiSgemvTop:XiDeconvTop:XiConvolutionTop --webtalk_flag SDSoC  --remote_ip_cache /home/ssssspe/FPGA/workspace/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" '
sds++ log file saved as /home/ssssspe/FPGA/workspace/filter2d/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

Best!
Jiaxing.

VishalX, 2018-07-25:
Please check #21.

closed 2018-07-25 by VishalX

bhargavajs07, 2018-10-14:
Hello @VishalX, I was trying to build the CHaiDNN framework targeted the XC7Z020 device(using ZedBoard) as the platform(using Xilinx SDx v2018.1) . And I get the same error as above : ps_e_S_AXI_HP*_FPD for in* cannot be found in the platform (and others as reported by the OP). 
The solution you suggest above of renaming the ports to zynq_ultra_ps_e_0_S_AXI_HP*_FPD seems to be specific to the ultrascale devices.

I see in the README that the framework is verified on zcu102 and zcu104 boards (Ultrascale) boards. Can you please tell if the framework could be modified in any way to get it to work on the ZedBoard ? 
(If so, can you please suggest appropriate port renaming I could use to fix this). I understand that there will be some additional resource constraints due to lack of UltraRAM in ZED Board can you also suggest any additional settings I might have to use to make it feasible to fit in.

Beprominent, 2018-12-18:
Hi, @HelloSongRen 
I am trying to import CHaiDNN to a xfopencv demo with HDMI input named stereo, and I want to build CHaiDNN to a shared library, but I encounter some problems. Did you include the example folder (i.e. ssd_ex.cpp) when you build the project? or just put folders (design and softward) into the revision project and build it?

Best|
Liangbin.

Beprominent, 2018-12-18:
Hi, @HelloSongRen 
I am trying to import CHaiDNN to a xfopencv demo with HDMI input named stereo, and I want to build CHaiDNN to a shared library, but I encounter some problems. Did you include the example folder (i.e. ssd_ex.cpp) when you build the project? or just put folders (design and softward) following the step about build chaidnn into the revision project and build it? 

Best!
Liangbin.

averr5, 2019-04-04:
@HelloSongRen In which design files did you change the port names? I am running into very similar errors while trying to build with zedboard.

HelloSongRen, 2019-04-04:
@averr5 
CHaiDNN/design/conv/include/xi_conv_config.h
#pragma SDS data sys_port(weights1:ps_e_S_AXI_HP0_FPD)

averr5, 2019-04-05:
@HelloSongRen Which device were you targeting ? I am getting errors relating to limited resources while targeting zedboards.

Mightyyy, 2022-01-29:
> Hi @HelloSongRen,
>
> HP port names are different in zcu102_rv_ss platform when compared to base platform. They need to be changed in the design files according to reVISION platform.
>
> Base platform HP [...]

Hi @VishalX I am facing the same issue as @HelloSongRen here. My platform is zc706. how should i rename the port names for the zc706 accordingly? thanks!