Xilinx/CHaiDNN #33: The CHaiDNN-v2 has worse performance?

pkurogerjs, 2018-07-24:
I  have built the hardware and sofrware code with Xilinx SDx 2018.2 tools successfully. But when I run the GoogleNet with 6-bit weights on ZCU102, the fps can only achieve 33.5, instead of the 220 fps presented in the documents.
The pre-built library works well, achieving a fps of 210 on ZCU102.
I found in the synthesis report that the project only uses about 444 DSPs ???
I am sure that I did not modify any codes in the repo. Could you please address this problem? Or is there anyone that has met the same problem?

VishalX, 2018-07-24:
Hi @pkurogerjs,

By default, the source files generate a 128 DSP @ 150/300 MHz core compute engine. That's why you are seeing a performance of ~33 FPS, which is expected. 

We have updated the config and Makefile to make the default configuration as 1024 DSP @ 200/400 MHz. As the base platform doesn't allow us to build design @ 250/500 MHz, a custom platform needs to be generated as mentioned in [CUSTOM_PLATFORM_GEN.md](https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md). 

Go through CHaiDNN documentation for more details on different configurations. Please pull the latest code and let me know if you still face any issue.

pkurogerjs, 2018-07-25:
Hello Vishao,
    Thank you for reply. I have found the configuration parameters before your reply and I adjusted the configs. But in that way I failed to build the whole project.
    It failed in the step "Phase 9 Depositing Routes" It reports that "Partial route confilts". I think it is because there is not enough resource.
   I saw the synthesis report that the convolution engine only uses 189 DPSs...

   Perhaps there is something wrong with the code?

jlamperez, 2018-07-25:
Hi @VishalX 

I an also having the above @pkurogerjs  error while compiling the bitstream. Here the complete log:

```
[23:51:11] Phase 4.3 Global Iteration 2
[00:44:43] Phase 5 Delay and Skew Optimization
[00:44:43] Phase 5.1 Delay CleanUp
[00:44:43] Phase 5.2 Clock Skew Optimization
[00:44:50] Phase 6 Post Hold Fix
[00:44:50] Phase 6.1 Hold Fix Iter
[00:44:56] Phase 6.2 Additional Hold Fix
[00:45:07] Phase 7 Route finalize
[00:45:07] Phase 8 Verifying routed nets
[00:45:13] Phase 9 Depositing Routes

===>The following messages were generated while  Compiling (bitstream) accelerator binary: bin Log file: /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/p0/vivado/prj/prj.runs/impl_1/runme.log  :
ERROR: [VPL RTSTAT-6] Partial route conflicts: 35212 net(s) have a partial conflict. The problem bus(es) and/or net(s) are zcu102_i/proc_sys_reset_5/U0/SEQ/ACTIVE_LOW_PR_OUT_DFF[0].FDRE_PER_N, zcu102_i/proc_sys_reset_3/U0/SEQ/ACTIVE_LOW_PR_OUT_DFF[0].FDRE_PER_N, zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/grp_ProcResult_fe_fu_1440/result_ping_fe0_3_V_U/a0_ProcResult_fe_resWhU_ram_U/ADDRARDADDR[4:0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/scale_buff_fb0_0_V_U/a0_ProcIStagingBuff_e9X_ram_U/ADDRARDADDR[0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff1_fd0_7_3_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff0_fd0_7_3_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/grp_ProcResult_fe_fu_1440/result_pong_fe0_3_V_U/a0_ProcResult_fe_resWhU_ram_U/ADDRARDADDR[4:0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff0_fd0_0_1_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[0], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff0_fd0_7_3_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[1], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff1_fd0_0_1_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[1], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/scale_buff_fb0_0_V_U/a0_ProcIStagingBuff_e9X_ram_U/ADDRARDADDR[1], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff0_fd0_0_1_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[1], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff1_fd0_7_3_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[1], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff1_fd0_0_1_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[2], zcu102_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1853/grp_ProcFeedingBuff_fz_fu_1084/grp_ProcInputBuff_fc_fu_709/grp_ProcWeightBuff_fd_fu_1535/weight_buff0_fd0_0_1_U/a0_ProcWeightBuff_fdbsm_ram_U/ADDRARDADDR[2]... and (the first 15 of 34622 listed), GLOBAL_LOGIC0.
ERROR: [VPL 12-1345] Error(s) found during DRC. Bitgen not run.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, write_bitstream ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2018.2/bin/vpl   --iprepo /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/iprepo/repo  --iprepo /opt/Xilinx/SDx/2018.2/data/ip/xilinx  --platform /opt/Xilinx/SDx/2018.2/platforms/zcu102/zcu102.xpfm  --temp_dir /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/p0  --output_dir /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/p0/vpl  --input_file /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/p0/.xsd/top.bd.tcl  --targe [... cut ...] CHaiDNN/design/build/../conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/jlamperez/Proyectos/CHaiDNN/design/build/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

```

Why is this happening?

I have also opened the vivado design to see the resources and I have seen for this configuration that the used DSPs are 1352. This value is closed to 1024 but is not exact, why is 1024 and where can I see it in the design?

[image]

VishalX, 2018-07-27:
@jlamperez 

Have you built this with base/default platform?
We have updated some documentation for building CHaiDNN @ higher frequencies which requires generation of a custom platform.

Please pull the latest code and go through the [README](https://github.com/Xilinx/CHaiDNN#run-inference) under section **Build CHaiDNN Hardware**. Also, to generate custom platform, follow the steps mentioned [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md).

jlamperez, 2018-07-27:
Hi @VishalX,

No, I created a custom platform based on your documentation. The clocks I added to this platform were 100Mhz and 200Mhz. Do I need to change them to 250Mhaz and 500Mhz to obtain 1024 DSPs?

cuongdv1, 2018-08-03:
Hi @VishalX,
I have question related to mAP,
How did you measure mAP of VGGSSD? 
I checked the accuracy of VGGSSD for two images camel.jpg and fish.jpg but it is too bad

I captured some results as below, confident_threshold = 0.25
-> for Camel.jpg
15 person 0.379514 597 1062 676 1305
9 chair 0.295174 524 894 618 1139
18 sofa 0.292740 835 933 937 1076
1 aeroplane 0.284229 111 255 213 379
3 bird 0.275507 133 243 221 350
15 person 0.270707 639 660 785 876
11 diningtable 0.268451 132 403 333 646
3 bird 0.258317 192 272 265 339
16 pottedplant 0.250684 1080 304 1166 503

-> for Fish.jpg
Detection num = 200
9 chair 0.353430 353 162 415 212
16 pottedplant 0.326995 339 205 419 232
2 bicycle 0.301073 345 223 416 273
15 person 0.285968 281 229 319 278
15 person 0.285354 309 227 342 264
9 chair 0.278175 327 196 367 226
9 chair 0.267440 248 96 316 160
15 person 0.266520 277 241 341 290
15 person 0.254333 309 223 377 269
15 person 0.253709 252 200 322 254

Please help 
Thank you very much!

VishalX, 2018-08-03:
@cuongdv1 

Your question is not really related to the issue in this thread. Could you please open this as another issue?

Thanks!

cuongdv1, 2018-08-03:
Sorry @VishalX 
I created new issue here #41 
please help me 

Best regards

VishalX, 2018-08-16:
@jlamperez Closing this issue. Please re-open if required.

closed 2018-08-16 by VishalX