Xilinx/CHaiDNN #34: How do I find output feature maps results and save to my disk?

Kang-An, 2018-07-24:
I want to know how to save the results of some layers, such as convolution layer's output feature maps and pooling layer?  
thanks.

AnushaPerla, 2018-07-27:
Hi Kang-An,
We had this feature but looks like it is broken. We will get back to you with this feature at the earliest opportunity.

NeutrinoXY, 2018-07-27:
See #25, 4th answer. It was for the v1, but it should work for the v2.

Kang-An, 2018-07-28:
Hi @NeutrinoXY , I've read your example code, but it has a little hard to understand
i know the `IN_PORTS `is 5, and the `OUT_PORTS `is 4. but what's difference in `in_ptr[0]`, `in_ptr[1]` ,`in_ptr[2]`, `in_ptr[3]`

can you tell me in detail how to get one channel feature map data in convolution? 

please, i'm waiting for your reply.

Kang-An, 2018-07-28:
@AnushaPerla 
I'm sorry to hear that and looking forward to your good news!

I think is's an important feature

AnushaPerla, 2018-08-01:
Hi @Kang-An 

We have added the feature of layer-wise output dump. Please pull the latest.

Refer to the [HW-SW PARTITIONING](https://github.com/Xilinx/CHaiDNN/blob/master/docs/HW_SW_PARTITIONING.md) doc to enable the feature. 

Thanks,
Anusha

closed 2018-08-01 by AnushaPerla