Xilinx/CHaiDNN #35: Access multiple bootom output results in Software Layer Plugin

Kang-An, 2018-07-25:
Hi, I've read the steps to create a custom software layer plugin, which says `"custom layer can take multiple bottom blobs, but only one top blob"`, The example code accesses a bottom result by `In_ptr 
 = (float*)xchange->in_ptrs[0]`, but how to access multiple bottom results?

By the way, can you introduce the differences between` in_ptr[0]`, `in_ptr[1]`, `in_ptr[2]`, `in_ptr[3]`, and `in_ptr[4]`

thank for your reply.

abid-k, 2018-07-26:
Hello @Kang-An,

You can access the first bottom result using `In_ptr0 = (float*)xchange->in_ptrs[0]` and second bottom result using `In_ptr1 = (float*)xchange->in_ptrs[2]`. Currently number of bottom blobs are limited to only 2 and only one top blob. Having said that we haven't tried any software plugins with more than 1 bottom yet.  Would this suffice for you? Could you please give it a try? 

We understand there is room for improvement in the interface. We will look into it and get back to you in the very next opportunity.

Can I please know more details about this layer you are trying to implement like number of bottom blobs, number of top blobs etc. ?

Kang-An, 2018-07-26:
Hi, @abid-k

I'm glad to see your reply. I plan to use CHaiDNN to implement Faster RCNN with feature pyramid, so I need to use 8 bottom blobs for connecting feature pyramid in fpn_propposal_layer.
finally, Score and bbox were obtained by using the results of the fpn_proposal_layer and roipool(for 4 bottom blobs)

I haven't tried to use `in_ptr[1]` to get a second bottom blobs because I don't know how to store the results of a particular convolution layer. I want to compare the result of convolution operation with the data pointed to by `in_ptr[1]`, and i can compare with the result of pure C version too

AnushaPerla, 2018-08-01:
@Kang-An

Refer to #34.

Kang-An, 2018-08-06:
Hi @AnushaPerla 
when I run some custom layers, I receive a error `[EP003] Current version supports one & only-one top blob`. in my prototxt, every layer has only one top,  Can you help me see what the problem is ? thank you

[FPN_Faster-RCNN.txt](https://github.com/Xilinx/CHaiDNN/files/2261568/FPN_Faster-RCNN.txt)

Kang-An, 2018-08-06:
Hi @AnushaPerla 
When i run a custom neural network, I got an error 
`[CHECKS] Start : Output Write

[WRPATH] out_path : models/FPN_Faster-RCNN/6Bit/out/rpn_cls_score_p5_out.txt
[CHECKS] (h, w, d, out_fbits, quant mode, th, sf) : 7 7 14 3 DynamicFixed 0 0

[CHECKS] End : Output Write

Seg fault @ 0x7faa798ab8

[CHECKS] Start : Output Write

[WRPATH] out_path : models/FPN_Faster-RCNN/6Bit/out/rpn_bbox_pred_p5_out.txt
lib/libxlnxdnn.so(+0x315d8)[0x7fb2ec65d8]
[CHECKS] (h, w, d, out_fbits, quant mode, th, sf) : 7 7 28 3 DynamicFixed 0 0

[CHECKS] End : Output Write

Seg fault @ 0x7fb0036008
lib/libxlnxdnn.so(+0x315d8)[0x7fb2ec65d8]`

Can you tell me how to solve it?
Here's my prototxt
[FPN_Faster-RCNN.txt](https://github.com/Xilinx/CHaiDNN/files/2261905/FPN_Faster-RCNN.txt)

Kang-An, 2018-08-08:
Hi @abid-k 
I tried to write a custom layer with two bottoms and a top, but i got this error:
`Currently only one top blob is supported for user_defined_layer: `

and i find it in `softwares/xtract.caffe_network_parser.cpp`

    string xlayerType = "XCustom";

    // Create new XLayer
    XLayer* dst = new XLayer(src.name(), xlayerType, src.top(0));
    
    // Assign actual layer type
    dst->xcustom_params->type = src.type();

    // Check if it is inPlace.
    int ntops = src.top_size();
    int nbottoms = src.bottom_size();
    ASSERT( (ntops > 0) && (nbottoms > 0), EP070, 
            "There should be minimum one top & bottom blobs for layer " << src.name())

    ASSERT( (nbottoms == 1), EP075, "Currently only one top blob is supported for user_defined_layer: " << src.name())

    ELOG( (ntops == 1 && nbottoms == 1 && src.bottom(0) == src.top(0)), EP073,"bottom and top blobs should be separate for user_defined_layer : " << src.name() )                        

    // Extract output dimensions [MANDATORY : Atleast one dimension should be given and it should be more than 0]
    ASSERT(src_parameter.top_dim_size() > 0, EP071,
            "Output dimension is not mentioned for user_defined_layer : " << src.name())

    dst->xcustom_params->output_dim.insert( dst->xcustom_params->output_dim.end(), 
                                            src_parameter.top_dim().begin(), src_parameter.top_dim().end());

    ASSERT(getSize(dst->xcustom_params->output_dim) > 0, EP072,
            "Total dimesion should be greater than 0 for user_defined_layer : " << src.name())

So, does this mean that the custom layer seems to support only one bottom blob?

abid-k, 2018-08-09:
Hi @Kang-An 

`ASSERT( (nbottoms == 1), EP075`  should have been `ASSERT( (ntops == 1), EP075` which is what the error msg says, "number of tops should be one".

Thank you for pointing it out.

I have changed it in repo. Could you pull the latest code and try it again?

Let us know if any issues.

Kang-An, 2018-08-09:
Hi @abid-k 
 i got a new error:
`[INFOx] Graph generated
[INFOx] Generating JobQueue
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted`

here's my custom layer and prototxt.
[FPN_Faster-RCNN-4-rpn_generator.txt](https://github.com/Xilinx/CHaiDNN/files/2273043/FPN_Faster-RCNN-4-rpn_generator.txt)

Can you tell me what went wrong？

    void Custom::custom_region_generator_1(const xChangeLayer *xchange){

	vector<int> out_dim_rpn = xchange->output_dims[0];	//	top blob
	vector<int> in_dim_bbox = xchange->input_dims[0];	//	bottom blob 1
	vector<int> in_dim_score = xchange->input_dims[1];	//	bottom blob 2

	float *In_ptr_bbox = (float*)xchange->in_ptrs[0];	//	bottom_data_bbox
	float *In_ptr_score = (float*)xchange->in_ptrs[1];	//	bottom_data_score
	float *out_ptr = (float*)xchange->out_ptrs[0];	//	top_data	rpn

	int in_size_bbox = 1;
	int in_size_score = 1;
	int out_size_rpn=1;
	int batch_bbox = in_dim_bbox[0];
	cout << "[Debug_resize] batch is: "<<batch_bbox<<endl;
	//	in_size = channel*width*height
	for(int it_num_in = 1; it_num_in < in_dim_bbox.size();it_num_in++){
		in_size_bbox = in_size_bbox * in_dim_bbox[it_num_in];
	}
	for(int it_num_in = 1; it_num_in < in_dim_score.size();it_num_in++){
		in_size_score = in_size_score * in_dim_score[it_num_in];
	}

	for(int it_num_out = 1; it_num_out < out_dim_rpn.size();it_num_out++){
		out_size_rpn = out_size_rpn * out_dim_rpn[it_num_out];

	}
	cout << "[Debug_resize] in_size_bbox_2 is: "<<in_size_bbox<< " in_size_score_2 is: "<<in_size_score<<" out_size_rpn is: "<<out_size_rpn<<endl;
	cout << "[Debug_resize] in_dim_bbox size is: "<<in_dim_bbox.size()<<" in_dim_score size is: "<<in_dim_score.size()<<" out_dim_rpn size is: "<<out_dim_rpn.size()<<endl;
	int channel = in_dim_score.at(1);	//	depth, channel	score
	int width_score = in_dim_score.at(2);	//	width_score
	int height_score = in_dim_score.at(3);	//	height_score

	int width_bbox = in_dim_bbox.at(2);	//	width_score
	int height_bbox = in_dim_bbox.at(3);	//	height_score
	cout << "[Debug_resize] channel_score is: " << channel << " width_score_out_2 is: "
		<< width_score << " height_score_out_2 is: " << height_score <<  endl;

	int spatial_size = width_score * height_score;
	int num = 0;

	float anchor_scale = 2;
	float anchor_stride = 4;

	vector<float> anchor_ratio;

	anchor_ratio.push_back(0.25);
	anchor_ratio.push_back(0.33);
	anchor_ratio.push_back(0.50);
	anchor_ratio.push_back(1.0);
	anchor_ratio.push_back(2.0);
	anchor_ratio.push_back(3.0);
	anchor_ratio.push_back(4.0);

	int imageWidth = 864;
	int imageHeight = 704;

	for(int batch_num = 0;batch_num < batch_bbox;batch_num++){
		
		out_ptr = (float*)xchange->out_ptrs[0];	//	top_data	rpn unknown length;
		int num = 0;
		
		Anchor anchors;
		AnchorData rpns;
		double temp = 0;
		nSize imageSize;
		imageSize.w = imageWidth;
		imageSize.h = imageHeight;
		for (int c = channel / 2; c < channel; c++){
		//reorg_cpu(In_ptr, w, h, c, 1, stride, forward, out_ptr);
			for(int i = 0; i < spatial_size; i++){
				
				int idx_h = i / height_score;	//	j
				int idx_w = i - idx_h * width_score;	//	r
				// Dtype *bottom_n = (Dtype*)(bottom_data + bottom[0]->offset(n,c));
				// Dtype *top_n = (Dtype*)(top_data + top[0]->offset(n,c));
				//cout << "[Debug_resize] channel status: "<<c<<" / "<<channel<<endl;
				float *In_ptr_bbox_0 = (float*)xchange->in_ptrs[0] + batch_num * ((c - 7)*4) * width_bbox * height_bbox;	//	bottom_data bbox[(i-7)*4][j][r]
				float *In_ptr_bbox_1 = (float*)xchange->in_ptrs[0] + batch_num * ((c - 7)*4+1) * width_bbox * height_bbox;	//	bottom_data
				float *In_ptr_bbox_2 = (float*)xchange->in_ptrs[0] + batch_num * ((c - 7)*4+2) * width_bbox * height_bbox;	//	bottom_data
				float *In_ptr_bbox_3 = (float*)xchange->in_ptrs[0] + batch_num * ((c - 7)*4+3) * width_bbox * height_bbox;	//	bottom_data
				In_ptr_score = (float*)xchange->in_ptrs[1] + batch_num * c * width_score * height_score;	//	bottom_data_score
				//In_ptr_bbox[idx_h * height_score + idx_w]	//bbox[(i-7)*4][j][r]
				
				anchors = anchor_generator((8+idx_w*anchor_stride),(8+idx_h*anchor_stride),anchor_scale,anchor_ratio[c-7]);

				rpns = rpn_generator((double)In_ptr_bbox_0[idx_h * height_score + idx_w],
									(double)In_ptr_bbox_1[idx_h * height_score + idx_w],
									(double)In_ptr_bbox_2[idx_h * height_score + idx_w],
									(double)In_ptr_bbox_3[idx_h * height_score + idx_w],
									anchors,imageSize);
				
				temp = min((rpns.xmax - rpns.xmin + 1.0), (rpns.ymax - rpns.ymin + 1.0));

				if(temp >= 6){
					out_ptr[num * 5 + 0] = rpns.xmin; 
					out_ptr[num * 5 + 1] = rpns.xmax;
					out_ptr[num * 5 + 2] = rpns.ymin;
					out_ptr[num * 5 + 3] = rpns.ymax;
					out_ptr[num * 5 + 4] = rpns.ymin;
					num += 1;
				}
			}
		}	
	}
}

abid-k, 2018-08-23:
Hi @Kang-An 

Do you still see this issue? If so, is it possible to share your caffemodel file also?

Kang-An, 2018-08-23:
Yes, the problem still exists. 

In fact, the layer defined in my caffemodel file needs up to 8 bottom blobs and 4 top blobs, because the custom layer of CHaiDNN only supports 2 bottom blobs and 1 top blob, so I have to split the layer into region_generator_1, region_generator_2, region_generator_3, region_generator_4...

But a few days ago I learned that the custom layer can support 30 bottom blobs. But it still only support one top blobs #51 

can you tell me how to make changes to make the program run on CHaiDNN?

This is my complete prototxt, which can be run with caffe and GPU.

the caffemodel is bigger than 10MB, so i can't upload  it, sorry.
[test_agnostic-resnet101-roi.txt](https://github.com/Xilinx/CHaiDNN/files/2313752/test_agnostic-resnet101-roi.txt)

abid-k, 2018-09-17:
Hi @Kang-An 

As mentioned in #51, currently we do not plan to support multiple top blobs. You can have upto 16 bottom blobs.

abid-k, 2018-09-17:
Closing this issue. Please re-open if required.

closed 2018-09-17 by abid-k