Xilinx/CHaiDNN #36: Can I use 2018.1 version of Vivado and Vivado_HLS?

hyungdal, 2018-07-27:
Hi,

My company has 2018.1 version of Vivado only.
I can't install 2018.2 version of Vivado in server because of permission.
Can I use 2018.1 version of Vivado for CHaiDNN?

Thank you.

Best Regards,
HyungDal.

VishalX, 2018-07-27:
@hyungdal,

2018.1 SDx doesn't support dual clocking feature which is required by CHaiDNN. I would suggest you migrate to the latest SDx tool (2018.2).

hyungdal, 2018-07-29:
> 보낸사람: "Vishal Jain" <notifications@github.com>
> 받는사람: "Xilinx/CHaiDNN" <CHaiDNN@noreply.github.com>
> 참조: "hyungdal" <feelpassion@naver.com>,"Mention" <mention@noreply.github.com>
> 날짜: 2018.07.27 오후 05:48:12

Thank you for your kind explanation.
I'll use 2018.2.

Thank you.

BR
Hyungdal

네이버 메일 앱에서 보냈습니다.

VishalX, 2018-07-30:
Closing. Please open new if you find issue with ChaiDNN.

Thanks!

closed 2018-07-30 by VishalX