Xilinx/CHaiDNN #37: ERROR: Linking globals named 'clock_start': symbol multiply defined!

salcanmor, 2018-07-27:
Hi guys,

I've seen #3 and #9 but those solutions did not solve the problem.

I have excluded from the build all the examples files from **src/software/example** except one and the error **ERROR: Linking globals named 'clock_start': symbol multiply defined!** is still there as you can see.

Any idea to fix this error?

`aarch64-linux-gnu-objcopy -O binary --set-section-flags .xdif=alloc --only-section=.xdif C:/CHaiDNN/CHaiDNN_v2/Release/src/design/conv/src/xi_convolution_top.o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.cdb/XiConvolutionTop_if.xml
C:/Xilinx/SDx/2018.2/bin/build_xd_ip_db -ip_search 0  -sds-pf C:/CHaiDNN/CHaiDNN_v2/Release/.Xil/zcu104.hpfm  -ip C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_PoolTop_1_0 -ip C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiConvolutionTop_1_0 -ip C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiDeconvTop_1_0  -o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.cdb/xd_ip_db.xml  
INFO: Using user-defined path for XILINX_XD environment variable C:/Xilinx/SDx/2018.2
processing accelerators: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_PoolTop_1_0
ip_dir: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_PoolTop_1_0
C:/Xilinx/SDx/2018.2/bin/xsltproc --stringparam xpath "spirit:component/spirit:name/text()" C:/Xilinx/SDx/2018.2/scripts/xdcc/xpathValueOf.xsl C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_PoolTop_1_0/component.xml
ip_name: PoolTop
processing accelerators: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiConvolutionTop_1_0
ip_dir: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiConvolutionTop_1_0
C:/Xilinx/SDx/2018.2/bin/xsltproc --stringparam xpath "spirit:component/spirit:name/text()" C:/Xilinx/SDx/2018.2/scripts/xdcc/xpathValueOf.xsl C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiConvolutionTop_1_0/component.xml
ip_name: XiConvolutionTop
processing accelerators: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiDeconvTop_1_0
ip_dir: C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiDeconvTop_1_0
C:/Xilinx/SDx/2018.2/bin/xsltproc --stringparam xpath "spirit:component/spirit:name/text()" C:/Xilinx/SDx/2018.2/scripts/xdcc/xpathValueOf.xsl C:/CHaiDNN/CHaiDNN_v2/Release/_sds/iprepo/repo/xilinx_com_hls_XiDeconvTop_1_0/component.xml
ip_name: XiDeconvTop
Generating data motion network
C:/Xilinx/SDx/2018.2/bin/llvm-link -o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/sds_all.o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe.pb.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe_network_parser.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xgraph.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xi_funcs.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract_opt.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract_utility.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_crop.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_fc_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_format_converter_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_nms_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_norm_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_permute_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_swsoftmax_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_perf_eval.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_scheduler.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_thread_routines.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_utils.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/interface/xi_interface.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/interface/xi_readwrite_util.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/init/xi_init.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/include/xchange_structs.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/imageread/xi_input_image.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/example/googlenet_ex.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/custom/custom_class.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/common/kernelinfo_class.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/common/xi_kernels.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/checkers/checkers.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/bufmgmt/xi_buf_mgmt.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/wrapper/dnn_wrapper.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/pool/src/pooling_layer_dp_2xio_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/deconv/src/xi_deconv_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/conv/src/xi_convolution_top.s

C:\CHaiDNN\CHaiDNN_v2\Release>C:\Xilinx\SDx\2018.2\llvm-clang\win64\llvm\bin\llvm-link.exe -o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/sds_all.o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe.pb.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe_network_parser.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xgraph.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xi_funcs.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract_opt.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract_utility.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_crop.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_fc_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_format_converter_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_nms_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_norm_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_permute_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/swkernels/xi_swsoftmax_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_perf_eval.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_scheduler.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_thread_routines.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/scheduler/xi_utils.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/interface/xi_interface.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/interface/xi_readwrite_util.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/init/xi_init.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/include/xchange_structs.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/imageread/xi_input_image.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/example/googlenet_ex.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/custom/custom_class.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/common/kernelinfo_class.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/common/xi_kernels.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/checkers/checkers.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/bufmgmt/xi_buf_mgmt.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/wrapper/dnn_wrapper.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/pool/src/pooling_layer_dp_2xio_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/deconv/src/xi_deconv_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/conv/src/xi_convolution_top.s 
ERROR: Linking globals named 'clock_start': symbol multiply defined!

C:\CHaiDNN\CHaiDNN_v2\Release>exit /b 1 
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling 'C:/Xilinx/SDx/2018.2/bin/llvm-link -o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/sds_all.o C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe.pb.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/caffe_network_parser.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xgraph.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xi_funcs.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract_opt.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/software/xtract/xtract [... cut ...] :/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/wrapper/dnn_wrapper.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/pool/src/pooling_layer_dp_2xio_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/deconv/src/xi_deconv_top.s C:/CHaiDNN/CHaiDNN_v2/Release/_sds/.llvm/./src/design/conv/src/xi_convolution_top.s'
sds++ log file saved as C:/CHaiDNN/CHaiDNN_v2/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed
sds++ completed at Fri Jul 27 12:08:10 +0200 2018
`

VishalX, 2018-07-27:
@salcanmor, 

Are you building using SDx GUI? If yes, please keep only one example file active in project while building and exclude other examples from the build.

salcanmor, 2018-07-27:
@VishalX that is exactly what I did and the error is still there

jlamperez, 2018-07-27:
If you see [here](https://github.com/Xilinx/CHaiDNN/search?q=clock_start&unscoped_q=clock_start) clock_start is defined in this files.

You can change the variable name instead of clock_start to clock_start_alexnet in the alexnet example

VishalX, 2018-07-27:
@salcanmor,

Could you comment lines 18 to 30 in [scheduler](https://github.com/Xilinx/CHaiDNN/blob/d5bf4142a76c583e7e60b7ec8bb0d4aff0dd31da/software/scheduler/xi_scheduler.cpp) file and try building. 

Let me know if that fixes your problem. We'll update the code.

VishalX, 2018-07-30:
Fixed. close #37
Please pull latest.

Thanks!

closed 2018-07-30 by VishalX