Xilinx/CHaiDNN #38: ERROR: slashes ( / ) are removed in some routes

salcanmor, 2018-07-30:
Hi guys,

I have tried to make a build following the CHaiDNN guide and I'm getting this error.

If you take a look to the error ( "C:CHaiDNNCHaiDNN_v2srcconvscriptsmcps.tcl"), you'll notice that all the "/" are removed.

Any idea to fix this error?

`===>The following messages were generated while  Compiling (top level synthesis checkpoint) dynamic region Log file: C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/p0/vivado/prj/prj.runs/synth_1/runme.log  :
ERROR: [VPL-1] couldn't read file "C:CHaiDNNCHaiDNN_v2srcconvscriptsmcps.tcl": no such file or directory
ERROR: [VPL 60-704] Integration error, One or more synthesis runs failed during dynamic region dcp generation
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling 'C:/Xilinx/SDx/2018.2/bin/vpl   --iprepo C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/iprepo/repo  --iprepo C:/Xilinx/SDx/2018.2/data/ip/xilinx  --platform C:/Xilinx/SDx/2018.2/platforms/zcu104/zcu104.xpfm  --temp_dir C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/p0  --output_dir C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/p0/vpl  --input_file C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:XiDeconvTop:XiConvolutionTop:adapter --webtalk_flag SDSoC  --remote_ip_cache C:/CHaiDNN/ip_cache --xp \"param:compiler.skipTimingCheckAndFrequencyScaling=1\" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1\" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={C:\CHaiDNN\CHaiDNN_v2\src\conv\scripts\mcps.tcl}\" --xp \"vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={C:\CHaiDNN\CHaiDNN_v2\src\conv\scripts\mcps.tcl}\" --xp \"param:compiler.deleteDefaultReportConfigs=false\" '
sds++ log file saved as C:/CHaiDNN/CHaiDNN_v2/Debug/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [CHaiDNN_v2.elf] Error 1

13:47:19 Build Finished (took 1h:4m:5s.907ms)`

VishalX, 2018-07-31:
Hi @salcanmor 

Looks like you are building ChaiDNN on Windows. 
If yes, there were cases where we've faced some tool related issues building projects on windows. The issue seems related to '/' on Linux machines vs. windows. Could you replace with windows style paths and try the build again?

salcanmor, 2018-07-31:
Hi @VishalX 

Yes, I'm building ChaiDNN on Windows. But it is being difficult.
I've solved that problem with this: #39 

But, when I solve one problem another one appears ...
I'm trying to solve all the problems.

Thank you.

VishalX, 2018-07-31:
@salcanmor 

We are reviewing your pull request. Once we confirm if that works for Linux SDx too, we will merge the changes.

Also, if you are having SDx-Windows specific problems, feel free to post on [Xilinx SDSoC Forums](https://forums.xilinx.com/t5/SDSoC-Environment-and-reVISION/bd-p/sdsoc).

VishalX, 2018-08-17:
@salcanmor,

We are not supporting CHaiDNN builds with SDx on Windows because of various tool related issue.
Closing this issue. 

If you are facing problems with SDx on Linux, please open a new issue.

closed 2018-08-17 by VishalX