Xilinx/CHaiDNN #40: ¿start/end layer in the graph1 & graph1?

salcanmor, 2018-07-31:
Hi guys,

I'm trying to run a new network. I've analyzed all the examples from /software/example/ and followed the "Running a Network using CHaiDNN" and there's a part which is missing in that guide.

In all the network examples from /software/example/ we can find this piece of code:

```
	//# start/end layer in the graph1
	string start_layer_graph1 = "";
	string end_layer_graph1   = "**pool5/7x7_s1**";

	//# start/end layer in the graph2
	string start_layer_graph2 = "**loss3/classifier**";
	string end_layer_graph2   = "";
```
Where the layers in bold change in every network. 
I'm trying to inference a new network and I don't know what to write in **end_layer_graph1** and **start_layer_graph2**.

How did you know which layers do you have to plave in those 2 variables?

Thanks in advance!

AnushaPerla, 2018-08-01:
Hello @salcanmor

`start_layer_graph1` is the name of the first layer of the network. If it is set to an empty string, CHaiDNN will take the first layer as in the prototxt by default.

Similarly `end_layer_graph1` is the name of the last layer of the network. If it is set to an empty string, CHaiDNN will take the last layer as in the prototxt by default.

The details are provided in the [API document](https://github.com/Xilinx/CHaiDNN/blob/master/docs/API.md). We have also updated the [RUN NEW NETWORK](https://github.com/Xilinx/CHaiDNN/blob/master/docs/RUN_NEW_NETWORK.md) doc also with these details.

CHaiDNN provides a feature to run a section of the network. In that case, you need to set these variables with corresponding first layer name and last layer name. You can look into [HW-SW-PARTITION](https://github.com/Xilinx/CHaiDNN/blob/master/docs/HW_SW_PARTITIONING.md) doc for more details on how you can use this feature for performance improvement.

AnushaPerla, 2018-08-20:
Closing this issue. Please re-open if required.

closed 2018-08-20 by AnushaPerla