Xilinx/CHaiDNN #41: the accuracy of VGGSSD on CHaiDNN-v2 is not good

cuongdv1, 2018-08-03:
I have question related to mAP,
How did you measure mAP of VGGSSD?
I checked the accuracy of VGGSSD for two images camel.jpg and fish.jpg but it is too bad

I captured some results as below, confident_threshold = 0.25
-> for Camel.jpg
//label-id, name, score, xmin, ymin, xmax, ymax
15 person 0.379514 597 1062 676 1305
9 chair 0.295174 524 894 618 1139
18 sofa 0.292740 835 933 937 1076
1 aeroplane 0.284229 111 255 213 379
3 bird 0.275507 133 243 221 350
15 person 0.270707 639 660 785 876
11 diningtable 0.268451 132 403 333 646
3 bird 0.258317 192 272 265 339
16 pottedplant 0.250684 1080 304 1166 503

-> for Fish.jpg
Detection num = 200
//label-id, name, score, xmin, ymin, xmax, ymax
9 chair 0.353430 353 162 415 212
16 pottedplant 0.326995 339 205 419 232
2 bicycle 0.301073 345 223 416 273
15 person 0.285968 281 229 319 278
15 person 0.285354 309 227 342 264
9 chair 0.278175 327 196 367 226
9 chair 0.267440 248 96 316 160
15 person 0.266520 277 241 341 290
15 person 0.254333 309 223 377 269
15 person 0.253709 252 200 322 254

Please help
Thank you very much!
[image]
[image]

AnushaPerla, 2018-08-03:
Hi @cuongdv1 

Could you please share more details about design configuration that you are using.
You can try to run with pre-built binaries [zcu102](https://github.com/Xilinx/CHaiDNN/tree/master/SD_Card/zcu102) or [zcu104](https://github.com/Xilinx/CHaiDNN/tree/master/SD_Card/zcu104)

Regarding mAP calculation, this [link](https://medium.com/@jonathan_hui/map-mean-average-precision-for-object-detection-45c121a31173) can be used as reference.

cuongdv1, 2018-08-04:
Hi @AnushaPerla 
which design configuration?
I cloned CHaiDNN.v2 source code from git and build project using SDx 2018.1, hardware platform zcu102
There is not any modification in the design configuration or ssd algorithm, I only insert some code to draw detected boxes run on zcu102

Thanks

cuongdv1, 2018-08-09:
Hi @AnushaPerla 
Thank you very much for your support
After install SDx 2018.2, i solved my problem

closed 2018-08-09 by cuongdv1

salcanmor, 2018-09-21:
>  There is not any modification in the design configuration or ssd algorithm, I only insert some code to draw detected boxes run on zcu102

Hi @cuongdv1 ,

Could you please share the code you use to draw the bounding boxes please?

Thanks in advance.