Xilinx/CHaiDNN #42: output of MobilenetSSD-300 on CHaiDNN is wrong

cuongdv1, 2018-08-03:
Hi guys
I am trying to convert MobilenetSSD-300 caffe model to CHaiDNN
I followed guidelines at https://github.com/Xilinx/CHaiDNN/blob/master/docs/RUN_NEW_NETWORK.md
But the detection output seems wrong as below.  Do you have any idea for this issue?

**test image: Camel.jpg** 
2 bicycle 1.000000 -407 1352 672 1648
2 bicycle 1.000000 1082 1336 1445 1812
2 bicycle 1.000000 720 1363 1624 1689
11 diningtable 1.000000 1126 1177 1395 1862
2 bicycle 0.500000 -64 189 627 331
2 bicycle 0.500000 876 1239 1690 1919
8 cat 0.500000 -64 189 627 331
16 pottedplant 0.500000 876 1239 1690 1919
2 bicycle 0.333333 -64 1318 304 1802
2 bicycle 0.333333 -208 1168 593 1871
10 cow 0.333333 -64 1318 304 1802
13 horse 0.333333 -64 1318 304 1802
13 horse 0.333333 -208 1168 593 1871
16 pottedplant 0.333333 -208 1168 593 1871

**test image: Fish.jpg** 
Detection num = 14
2 bicycle 1.000000 -150 273 248 332
2 bicycle 1.000000 401 268 533 365
2 bicycle 1.000000 266 275 601 341
11 diningtable 1.000000 416 237 516 375
2 bicycle 0.500000 -23 39 232 67
2 bicycle 0.500000 320 250 626 387
8 cat 0.500000 -23 39 232 67
16 pottedplant 0.500000 320 250 626 387
2 bicycle 0.333333 -24 266 113 363
2 bicycle 0.333333 -75 235 221 377
10 cow 0.333333 -24 266 113 363
13 horse 0.333333 -24 266 113 363
13 horse 0.333333 -75 235 221 377
16 pottedplant 0.333333 -75 235 221 377

Please help
Thanks in advance!

VishalX, 2018-08-07:
@cuongdv1 
Could you share more details on design config, device, platform?
Also, please share the details of quantization scheme you've used.

VishalX, 2018-08-10:
Hi @cuongdv1 

We haven't verified MobileNetSSD yet. If you could share the prototxt and caffemodel or point us to the source, if possible, and we will have a look.

cuongdv1, 2018-08-11:
Hi @VishalX 

Attached file is quantized prototxt file, 
Original prototxt and caffemodel I cloned at here (https://github.com/chuanqi305/MobileNet-SSD)
Please review quantized prototxt and teach me if you found any problem

Thank you very much

[MobilenetSSD-300.txt](https://github.com/Xilinx/CHaiDNN/files/2279885/MobilenetSSD-300.txt)

VishalX, 2018-10-11:
Duplicate of #53.
Please use other thread to report issues regarding MobileNetSSD. Closing this.

closed 2018-10-11 by VishalX

adu81020799, 2018-10-30:
Hello 
could you please share your makefile which you used for compiling your elf file?

Regards
Adarsh Sawant