Xilinx/CHaiDNN #43: CHaiDNN on other boards

MiguaMG, 2018-08-04:
Dear all,

Has anyone tested CHaiDNN on other type of boards besides zcu102 and zcu104?
In particular I'm thinking on the possibility of using the light version of CHaiDNN (Diet CHaI) on low cost boards (such as the Zybo, Pynq ...). The idea would be using CNNs with very few layers so presumably there should be enough hardware to handle that.

nkpavan, 2018-08-07:
Hi @MiguaMG ,
We have not tried building CHaiDNN for Zybo or Pynq boards. You can try building it for low cost boards,  given the device you are targeting has sufficient resources to fit DietCHai. You can find the DietCHai resource utilization [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/PERFORMANCE_SNAPSHOT.md).

Dhananjayadmd, 2018-08-20:
Even ZC702 board does not have sufficient resource for DietCHai. Do you have any plan to downscale the design further and give support for ZC702 board?

trihook, 2018-08-20:
@nkpavan Does DietCHai was supported CHaiDNN v2 ?

bhargavajs07, 2018-10-14:
Hello @Dhananjayadmd, I too was trying to build the CHaiDNN framework (in DietCHai mode by mapping only the conv layer to the HW accel, Pooling and DeCONV in S.W)  targeted the XC7Z020 device(using ZedBoard) as the platform(using Xilinx SDx v2018.1) . 

And I got the an error : ps_e_S_AXI_HP*_FPD for in* cannot be found in the platform ( similar to the one reported in #30  by the OP).
The solution you suggested there, was to rename the ports to zynq_ultra_ps_e_0_S_AXI_HP*_FPD seems to be specific to the ultrascale devices.

Could you please tell if you too came across similar issue, if so, how did you fix it ?

VishalX, 2018-10-29:
@Dhananjayadmd @MiguaMG,

Please refer to issue #93.

VishalX, 2018-11-02:
Closing. Reopen if required.

closed 2018-11-02 by VishalX