Xilinx/CHaiDNN #46: Custom machine, Integration error, problem implementing dynamic region, route_design ERROR

jlamperez, 2018-08-14:
Hi,

I have seen in issue #12, that this problem is fixed for the new CHaiDNN v2 release but when I tried to use a custom machine I am having it again. 

My custom machine has a clock wizard with this frequencies:

[image]

When I compile the Makefile with 100Mhz clock everything compiles fine and I am able to run different networks on the board but Integration error occurs when I try to compile for 200Mhz.

Can you tell me @VishalX how did you fix it? 

Also with this clocks when I try to compile for 250Mhz I get the next error:

```
ERROR: [CF2XD 83-23150] No derived clock for platform 'SK0808_zusys_SDSoC' clock domain 'SK0808_zusys_SDSoC_clk_wiz_0_0_clk_out1' for clock id 3 with clock period multiplier = 1 and clock period divisor = 2
WARNING: [CF2XD 83-1006] Invalid clock id -1 for platform SK0808_zusys_SDSoC
ERROR: [CF2XD 83-1392] Illegal port name 'INVALID_ID' for comp 'SK0808_zusys_SDSoC'
ERROR: [CF2XD 83-1307] getPort(INVALID_ID, SK0808_zusys_SDSoC) is null
```

Any idea of why is this clock invalid if I have it in the clock wizard?

Thanks,

Jorge

nkpavan, 2018-08-20:
Hi @jlamperez ,

For meeting timing constraints at 250/500 MHz, follow the steps mentioned in Issue #48 .

closed 2018-08-22 by jlamperez

lishen565, 2019-01-31:
Hi,@jlamperez , I also meet the timing constraints at250/500MHz. The custom platform works well at 400/200Mhz,which guide by https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md. But after I tried to Iterative synthesis and P&R strategy as https://github.com/Xilinx/CHaiDNN/issues/48, I met  the timing constraints problem.Could you tell me how do you solve it? Thanks a lot.

vietthanh85, 2019-04-10:
Hi @nkpavan , I also met the same problem as @lishen565's problem above . Could you suggest some solutions to solve it? Thaiks a lot.