Xilinx/CHaiDNN #47: mcp_const.xdc, support TE0808-04-09-1EE-S StarterKit

jlamperez, 2018-08-14:
Hi,

I reference the issues #43 and #46 because they can be related.

I am building CHaiDNN v2 for [TE0808-04-09-1EE-S](https://shop.trenz-electronic.de/en/TE0808-04-09-1EE-S-TE0808-04-09-1EE-S-Starter-Kit) Starter kit which contains a ZU9EG module.

I have created a custom SDSoC machine for the board and I am able to run CHaiDNN executables (GoogleNet, Alexnet, Resnet and SSD) @100/200Mhz, 1024DSP with the next performance:

* Alexnet

```
./alexnet.elf Xilinx 6 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 5.852938 Images/second
```
./alexnet.elf Xilinx 8 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 5.984368 Images/second

* GoogleNet_V1_NoLRN

```
./googlenet.elf Xilinx 6 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 92.024878 Images/second
```
./googlenet.elf Xilinx 8 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 62.667814 Images/second

*  ResNet-50-BNfused

```
./resnet50.elf Xilinx 6 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 29.733502 Images/second
```
./resnet50.elf Xilinx 8 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 20.563600 Images/second

* VGG-SSD300 
```
./ssd.elf Xilinx 6 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 3.027429 Images/second
```
./ssd.elf Xilinx 8 images/camel.jpg images/goldfish.JPEG
```
>[PERFM] Performance with Batching : 2.286070 Images/second

Now I would like to get a better performance and approximate to the ones listed in the [performance tables](https://github.com/Xilinx/CHaiDNN/blob/master/docs/PERFORMANCE_SNAPSHOT.md) but when I try to compile for another clock frequencies I am different problems.

To try to solve this problems I have opened the created vivado.log and there are some points that I have not very clear:

1.- When SDSoC sources mcp_const.tcl I have this Critical Warnings

```
CRITICAL WARNING: [Vivado 12-4739] set_multicycle_path:No valid object(s) found for '-from [get_cells -hierarchical -filter { NAME =~  "*din6_buf*" && PRIMITIVE_TYPE =~ REGISTER.*.* }]'. [/home/jlamperez/Proyectos/CHaiDNN_250_500/design/conv/scripts/mcp_const.xdc:6]
Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.
WARNING: [Vivado 12-508] No pins matched 'zcu102_i/clk_wiz_0/inst/plle4_adv_inst/CLKOUT0'. [/home/jlamperez/Proyectos/CHaiDNN_250_500/design/conv/scripts/mcp_const.xdc:7]
INFO: [Timing 38-35] Done setting XDC timing constraints. [/home/jlamperez/Proyectos/CHaiDNN_250_500/design/conv/scripts/mcp_const.xdc:7]
WARNING: [Vivado 12-1008] No clocks found for command 'get_clocks -of_objects [get_pins zcu102_i/clk_wiz_0/inst/plle4_adv_inst/CLKOUT0]'. [/home/jlamperez/Proyectos/CHaiDNN_250_500/design/conv/scripts/mcp_const.xdc:7]
Resolution: Verify the create_clock command was called to create the clock object before it is referenced.
```

So, in  [mcp_const.xdc](https://github.com/Xilinx/CHaiDNN/blob/d5bf4142a76c583e7e60b7ec8bb0d4aff0dd31da/design/conv/scripts/mcp_const.xdc) I am seeing that it references to zcu102_i, do I need to modify this constraint file? Why is working @100/200 if I haven't modify this file for the starter kit?

2.- Can this damage the board?

```
CRITICAL WARNING: [Vivado 12-4430] You are lowering the severity from an ERROR. This can potentially lead to board damage or other functionality issues on hardware. If this is not desired, please run 'reset_property SEVERITY [get_drc_checks HDPR-5]' to undo this change.
```

3.- This is the one explained in issue #46 

```
problem implementing dynamic region, route_design ERROR
    while executing
"error {problem implementing dynamic region, route_design ERROR}"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 error [list $msg]"
    (procedure "error2file" line 12)
    invoked from within
"error2file $cwd "problem implementing dynamic region, $run_status" "
    (procedure "ocl_util::single_project_flow_unip" line 234)
    invoked from within
"ocl_util::single_project_flow_unip $dsa_info $config_info $clk_info $debug_profile_info"
    (file "/home/jlamperez/Proyectos/CHaiDNN_250_500/design/build/_sds/p0/vivado/ipirun.tcl" line 207)
```

Can someone help me debugging this? I can share custom SDSoC platform if needed.

I would like to write some tutorials about this work once the platform is more stable.

Thanks,

Jorge

jlamperez, 2018-08-21:
So if I use CHaiDNN by default for zcu102 CLOCK_ID :=3 (200Mhz) and DM_CLOCK_ID:=3.

This machine has this clocks:

[image]

If the clocks in the schematic are this ones:

[image]

I  need to change `zcu102_i/clk_wiz_0/inst/plle4_adv_inst/CLKOUT0 with zcu102_i/clk_wiz_0/inst/mmcme4_adv_inst/CLKOUT5 (400Mhz)` and `zcu102_i/clk_wiz_0/inst/BUFGCE_DIV_CLK2_inst/O with zcu102_i/clk_wiz_0/inst/clkout4_buf/O (200Mhz) ` right?

nkpavan, 2018-08-24:
That's right @jlamperez . However for better timing you can configure the clock wizard IP as described [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md).

jlamperez, 2018-08-28:
Hi @nkpavan 

Okay, I have tried compiling the project for the zcu102 again as described in the documentation ( with 2 clocks) and I can see that I don't need to change the *mcp_const.xdc* because the clock now is a pll instead of mmcme and there isn't any timing error.

So I have followed the same behaviour for the  TE0808-04-09-1EE-S StarterKit but in this case after compiling the design I am seeing some timing errors. I have added some extra IP Cores to the design so maybe this why I am seeing them. Can you give me some advice to minimize this timing errors?

Thank you very much,

jlamperez, 2018-08-29:
Sorry, I mean that timing constraints are not being met in the design.

nkpavan, 2018-08-29:
CHaiDNN (1024 DSP config) utilizes ~90% BRAMs on zcu102. It will be difficult to meet timing if other IPs are added. Could you share the details of the other IPs you added and their resource utilization and also what frequency are you targeting?

jlamperez, 2018-08-30:
Hi @nkpavan

Here a picture of the IPs, in yellow are extra added IP cores.

[image]

 The frequency is 50 and 100 Mhz

nkpavan, 2018-09-05:
Hi @jlamperez ,

Could you try with a smaller configuration (256 or 512 DSPs). You can find the configurable parameters [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/CONFIGURABLE_PARAMS.md)

closed 2018-10-11 by nkpavan