Xilinx/CHaiDNN #48: Iterative synthesis and P&R strategy

jlamperez, 2018-08-14:
Can somebody point me out with some resources of how can I do this iterative strategy?

Do I need to open Vivado design and make it? How is created libxlnxdnn.so if I build from Vivado and with the hardware Makefile?

Thanks!

nkpavan, 2018-08-20:
Hi @jlamperez ,

Yes, you need to open the design using Vivado for iterative strategy as mentioned in the below steps. 

1.	Build CHaiDNN using Makefile flow for 200/400 MHz using the custom platform generated as mentioned [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md).
2.	After the build is complete, open the Vivado project (.xpr) and change the clock frequencies 400 and    200 MHz to 500 and 250 respectively.
3.	Re-launch the implementation phase.
4.	After the implementation is complete, generate the bit-stream.
5.	Copy the bit-stream to build folder and rename it to libxlnxdnn.so.bit
6.	Run the below command in shell prompt to generate the BOOT.BIN file
bootgen –arch zynqmp –image boot.bif –w –o i BOOT.BIN
7.	Copy the generated BOOT.BIN and replace it with the one present in the sd_card folder

jlamperez, 2018-08-22:
Hi @nkpavan ,

Thank you for your answer, I have followed the steps and I was able to create a BOOT.BIN for 250/500 Mhz. I will compare my results with the ones in [performance/utilization](https://github.com/Xilinx/CHaiDNN/blob/master/docs/PERFORMANCE_SNAPSHOT.md)

Closing!

closed 2018-08-22 by jlamperez