Xilinx/CHaiDNN #50: Support for large image sizes

takayuki-kita, 2018-08-20:
Hi,

I think most of the prototxt of the sample corresponds to the image size of 224 x 224. 
```
name: "VGG_ILSVRC_16_layers"
layer {
  name: "data"
  type: "Input"
  top: "data"
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 224
      dim: 224
    }
  }
}
```
I am experimenting with a network with a larger size (eg 480 x 600).
```
layer {
  name: "data"
  type: "Input"
  top: "data"
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 480
      dim: 600
    }
  }
}
```
However, when executed on this network, there is no response from PoolTop.

The definition of Pooling seems to be limited to 480 x 480. 
- design/pool/include/maxpool_config.h
```
#define OUTWIDTH	　480
#define OUTHEIGHT　　　　　480
```

So, I changed the OUTWIDTH and OUTHEIGHT, but it was the same result.
```
#define OUTWIDTH	　480*2
#define OUTHEIGHT　　　　　480*2
```
Is it not enough to make the above changes to correspond to a large image size?
Or is there a mistake in correspondence?

Thanks.

VishalX, 2018-08-20:
@takayuki-kita 

Could you please share more information on the build configuration/Device/frequency etc?
Also, Please share your prototxt if possible.

VishalX, 2018-08-20:
@takayuki-kita 
Thanks for sharing the details. I have some more questions before I look further.

・224x224 is OK.
・300x300 is OK.
・200x480 is OK.
> Does it mean that these networks are working functionally and are able to classify images correctly?

・200x600 is NG. (PoolTop does not respond)
・480x600 is NG. (PoolTop does not respond)
> Can you be more specific about the issue regarding pool (PoolTop does not respond)? Does the HW stall while executing pool or something else? 
> How did you come to the conclusion that Pool is having the issue? Did you enable any messages in the scheduler? Can you share the log if you have any?

Thanks

takayuki-kita, 2018-08-21:
As a supplement, my environment of this issue is generated with [Fix for 8KP design](https://github.com/Xilinx/CHaiDNN/commit/84318fb2ac96e19bebcc62fd8574ab041d28baac) imported.

After importing the latest code, Another problem occurred. (Issue#52 [Can't compile latest version](https://github.com/Xilinx/CHaiDNN/issues/52))

VishalX, 2018-08-21:
@takayuki-kita 

We use async calls to hardware accelerated functions and ENABLE_CONSOLE_TEXT prints the messages sequentially. Hence, doesn't give much idea where the execution stalls/waits.

Are you able to kill the application with ctrl+c or the board becomes unresponsive?
If yes, it's not a hardware hang & there is something wrong with the software. 

Can you verify your tests with a build using Makefile and not with GUI? If you see similar behavior, we can dig this further. Meanwhile, we will try to replicate your issue.

takayuki-kita, 2018-08-21:
Hi, @VishalX

> Are you able to kill the application with ctrl+c or the board becomes unresponsive?

takayuki-kita, 2018-08-29:
Hi, @VishalX 

I compiled with makefile, but the results were similar. 
I want to know where to change when increasing the height or width of the image size. 

Thanks.

VishalX, 2018-11-02:
@takayuki-kita,

You might need to check if the data fits inside the BRAM storage for a specific case. In the pool case, we have limited BRAMs. Pool needs some data to be present inside BRAMs meet minimum processing requirements. If that doesn't meet, you might encounter unknown behavior/results.

We'll try to put some run time warnings, that would highlight some of these cases, in the next release.

takayuki-kita, 2018-11-14:
Hi, @VishalX

Sorry for the late reply. 
I will try to check if the data fits inside the BRAM storage in the pool case.
And I am also waiting for warning implementation. 

It is OK to close this Issue.
Thank you.

VishalX, 2018-11-19:
Thanks! @takayuki-kita 

Closing.

closed 2018-11-19 by VishalX