Xilinx/CHaiDNN #51: custom layer execution

Kang-An, 2018-08-21:
Hi 

I already know that CHaiDNN's custom layer only supports two bottom blobs and one top blob, so I save the remaining bottom blob to global memory for custom layer calls. 

For Example: the custom layer requires the calculation results of conv1, conv2, conv3, conv4. In prototxt, I pass conv1 and conv2 to the custom layer, and the results of conv3 and conv4 are saved in the global memory or file. When the custom layer is executed, the layer accesses conv1 and conv2 from memory and conv3 and conv4 from global memory or files.

Since CHaiDNN uses hardware and software partitioning and multi-thread parallel execution, **how can I be sure that the calculation results of conv3 and conv4 have been saved to global memory during the custom layer execution**?

I tried to run it, but it seems that the hardware acceleration part has not yet calculated conv3 and conv4 when executing the custom layer. So I got a wrong calculation result

AnushaPerla, 2018-08-21:
Hi @Kang-An 

We have recently increased the number of bottom blobs support to 16. The 16 bottom blobs will be available at these positions "in_ptrs[0], in_ptrs[2],... ,in_ptrs[30]"  in xChangeLayer structure. Refer to [xchange_structs.cpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/include/xchange_structs.hpp) for more details.

Could you please checkout latest version and try.

Kang-An, 2018-08-21:
Hi @AnushaPerla 

It's good to hear this news. can you tell me the maximum number of top blobs?  still one top blob?

AnushaPerla, 2018-08-23:
only one top blob is supported

Kang-An, 2018-08-23:
thanks, Can I ask if you have plans to expand the number of top blobs? My layer needs multiple top blobs..

AnushaPerla, 2018-08-27:
Hi @Kang-An,

Currently we do not plan to support multiple top blobs. As mentioned by you in #35, Could you please try to split the layer into four layers.

Kang-An, 2018-08-27:
ok, thank you for your reply. @AnushaPerla