Xilinx/CHaiDNN #52: Can't compile latest version.

takayuki-kita, 2018-08-21:
Hi,

I checked out latest version. ([8 KERNEL PROC Changes](https://github.com/Xilinx/CHaiDNN/commit/2ac2f2fc76e99ec3bdea7f2ccac926c196962cc1))

But I couldn't compile it using SDx IDE.
My Platform is zcu102.

An error related to arm_neon occurs. 
```
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:31:13: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma GCC target ("+nothing+simd")
            ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:38:9: error: unknown type name '__Int8x8_t'
typedef __Int8x8_t int8x8_t;
        ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:39:9: error: unknown type name '__Int16x4_t'
typedef __Int16x4_t int16x4_t;
        ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:40:9: error: unknown type name '__Int32x2_t'; did you mean '__int32_t'?
typedef __Int32x2_t int32x2_t;
 :
 :
```

I thought that it was necessary to specify "-mfpu = neon" in the compiler option,
I don't know how to specify with the SDSOC ++ compiler.

Or do I have to take another countermeasure?
Please tell me how to resolve.

e1126925, 2018-08-27:
Hi,

I'm having the same problem. Did you find any solution yet.

takayuki-kita, 2018-08-28:
Hi,

I could compile using makefile without IDE. 
In case of IDE, some option specification may be missing.

e1126925, 2018-08-28:
Ok thank you then I will try that first. I hope somebody has a solution for the IDE aswell

e1126925, 2018-08-29:
ok i've got a little further now. i noticed that in the guide for the SDx build process the Symbold __SDSOC__ is missing the 2 underscores at the end (at least i think that's how it should be). But now i'm facing another problem

ERROR: [SYNCHK 200-61] /home/data/Xilinx/zcu102-rv-ss-2018-2/workspaces/CHaiDNN_2/CHaiDNN/src/design/deconv/src/xi_deconv.hpp:388: unsupported memory access on variable 'scalar_deconv_args' which is (or contains) an array with unknown size at compile time.
ERROR: [SYNCHK 200-22] /home/data/Xilinx/zcu102-rv-ss-2018-2/workspaces/CHaiDNN_2/CHaiDNN/src/design/deconv/src/xi_deconv.hpp:107: memory copy is not supported unless used on bus interface possible cause(s): non-static/non-constant local array with initialization).
WARNING: [SYNCHK 200-23] /home/data/Xilinx/zcu102-rv-ss-2018-2/workspaces/CHaiDNN_2/CHaiDNN/src/design/deconv/src/xi_deconv.hpp:158: variable-indexed range selection may cause suboptimal QoR.
INFO: [SYNCHK 200-10] 2 error(s), 1 warning(s).

xiaokaijia1988, 2018-09-03:
I'm having the same problem. Did you find any solution yet.

xiaokaijia1988, 2018-09-04:
@takayuki-kita Did you find any problem under IDE?

e1126925, 2018-09-05:
Unfortunately no. I tried fixing the error but it runs into a similar error right afterwards

takayuki-kita, 2018-09-05:
Problems when using IDE, I have not solved it yet. I will report it when a solution is found.

salcanmor, 2018-09-11:
another one here is the same problem, @VishalX do you have any solution for this?
By the way, do we have to add  __SDSOC or __SDSOC__ ???

VishalX, 2018-09-11:
The symbol is `__SDSOC`. There is no need to add trailing underscores.

@takayuki-kita, Is the original issue reported here solved? Regarding ARM compilation error?

salcanmor, 2018-09-11:
@VishalX you placed some code in 52#issuecomment-414640833. 

I'm not using others opencv functions, I'm just trying to build the project from scratch, according to the [guide](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md).

So, if that (52#issuecomment-414640833) is a valid solution for my case, can you please give us a brief explanation on where to place that code?

salcanmor, 2018-09-11:
@VishalX,

Can you please tell me how to build the project using Makefile ?

Thanks in advance.

VishalX, 2018-09-11:
Please follow the **Build from Source** section in [README](https://github.com/Xilinx/CHaiDNN/blob/master/README.md).

salcanmor, 2018-09-11:
@VishalX,

Is is possible to have that guide for ZCU104 ?

Thanks in advance.

nkpavan, 2018-09-12:
Hi @salcanmor,

To build CHaiDNN for zcu104, some of the storage structures in the design needs to be mapped to URAMs. Also the standalone pool and deconv kernels needs to be disabled (pool functionality is also implemented in conv kernel). For detailed steps refer Issue #56 .

salcanmor, 2018-09-12:
@VishalX @nkpavan 

All the info to build CHaiDNN in ZCU104 is scattered and incomplete.
Since, CHaiDNN-v2 cannot be build in SDx 2018.2, is it possible to make a guide to make the build from source?

Thanks in advance.

VishalX, 2018-10-08:
Please check Issue #60 for arm_neon related issue.

VishalX, 2018-10-08:
#60

VishalX, 2018-10-11:
Duplicate of #60

closed 2018-10-11 by VishalX