Xilinx/CHaiDNN #53: MobileNet-SSD run on CHaiDNN v2 fail.

tuananhcoder, 2018-08-22:
Hi,
I quantize model deloy of Mobilenet-SSD on https://github.com/chuanqi305/MobileNet-SSD
follow Xilinx quantization on this guide: https://github.com/Xilinx/CHaiDNN/blob/master/docs/QUANTIZATION.md
But run failure. 
MobileNet-SSD input image pixel is on range [-1:1],
so I set run file quantize is:

```
./chaidnn_tools_ENV/bin/python2 XportDNN.pyc --quant_type "Xilinx" \
--deploy_model /home/anhnt/CHaiDNN/tools/tools/MobileNet-SSD/deploy.prototxt \
--weights /home/anhnt/CHaiDNN/tools/tools/MobileNet-SSD/MobileNetSSD_deploy.caffemodel \
--quantized_deploy_model /home/anhnt/CHaiDNN/tools/tools/MobileNet-SSD/models_quantd/MobileNetSSD_deploy_quantd.prototxt \
--calibration_directory /home/anhnt/CHaiDNN/tools/tools/JPEGImages/ \
--calibration_size 32 \
--bitwidths 8,8,8 --dims 3,300,300 --transpose 2,0,1 \
--channel_swap 2,1,0 --raw_scale 0.007843 \
--mean_value 1,1,1 --input_scale 1.0
```

but got error( hang run):
```
/usr/local/lib/python2.7/dist-packages/skimage/io/_io.py:49: UserWarning: `as_grey` has been deprecated in favor of `as_gray`
  warn('`as_grey` has been deprecated in favor of `as_gray`')
/usr/local/lib/python2.7/dist-packages/skimage/transform/_warps.py:105: UserWarning: The default mode, 'constant', will be changed to 'reflect' in skimage 0.15.
  warn("The default mode, 'constant', will be changed to 'reflect' in "
/usr/local/lib/python2.7/dist-packages/skimage/transform/_warps.py:110: UserWarning: Anti-aliasing will be enabled by default in skimage 0.15 to avoid aliasing artifacts when down-sampling images.
  warn("Anti-aliasing will be enabled by default in skimage 0.15 to "
quantize.py:33: RuntimeWarning: invalid value encountered in divide
  #     pass
quantize.py:33: RuntimeWarning: divide by zero encountered in log2
  #     pass
```
I quantized with params same the guide, but got accuracy very bad.
Please help me,
Thank you!

smenon009, 2018-08-22:
Hi @tuananhcoder,
Great to know that you are able to follow the quantization guide. 

MobileNetSSD mentioned in the link provided uses mean as 127.5 for the all the input channels and has a input scale of 0.007843. So, could you try changing the --input_scale to 0.007843 and leave the --raw_scale with the default? 

Did you get very bad accuracy when you successfully ran the quantizer with default settings?

tuananhcoder, 2018-08-23:
Hi @smenon009 
Thank you for you comment!

> Did you get very bad accuracy when you successfully ran the quantizer with default settings?
Yeah I ran the quantizer with default settings and got fail in detection.
I have been ran the quantizer with mean_value=127.5, raw_scale=255.0(default raw_scale=255.0) and input_scale=0.007843 but got fail in run detection.

VishalX, 2018-10-29:
@tuananhcoder,

We are going to support MobileNetSSD with our next release which will probably come out in Mid-December. It will be available as a part of ChaiDNN-ModelZoo.

closed 2018-11-02 by VishalX

cuongdv1, 2018-11-05:
@VishalX 
I want to port MobilenetSSD on CHaiDNN by myself
Currently which part of MobilenetSSD was not supported?
Could you tell me the part that need to be modified to support MobilenetSSD model?

Thank you very much in advance

titikid, 2018-12-05:
@VishalX is there any update information about MobilenetSSD? 
btw, Can you share the quantization flow of "Xilinx" mode? is it similar with tensorRT flow [here](http://on-demand.gputechconf.com/gtc/2017/presentation/s7310-8-bit-inference-with-tensorrt.pdf)?
Thank you.

xlxxxxxl, 2019-07-02:
Hi @cuongdv1 :
Have you solved the mobilenetSSD problem now? How did you solve it?
Do you know why CHAIDNN does not update it?