Xilinx/CHaiDNN #54: TE0808-04-09-1EE-S Performance comparison

jlamperez, 2018-08-22:
Hi,

Based in #48 issue I have created a design for 250/500 Mhz but although I see improvements regarding the 200/400 Mhz design there are some differences with the ones publicly available.

Here a table for  TE0808-04-09-1EE-S Performance in fps using the models from the [Model Zoo](https://github.com/Xilinx/CHaiDNN/blob/master/docs/MODELZOO.md)

| Network | 6-bit | 8-bit |
|   ------     |  ----- | ------ |
| GoogleNet_V1_LRN |  199.76 | 141.02 |
| Alexnet |  6.06 | 6.03  |
| ResNet | 57.94 | 40.22 |
| ALEXNET-FCN | 10.66  |  9.60|

* For VGG-SSD300 I am having the next error when I run `./ssd.elf Xilinx 6 images/camel.jpg images/goldfish.JPEG` : 

```
[INFOx] Network Path : models/VGG_SSD_300/8Bit

[INFOx] Init Start : This may take a while ...
[INFOx] Design will not fit for input dimension  256x1x1 with filter dimension 3x3
```

Any idea of why?

Comparing performances with the ZU, 

[image]

is there any possibility for the TE0808-04-09-1EE to get closer to this values? (For example meeting the timing constraints or something else?)

Thanks,

Jorge

jlamperez, 2018-08-27:
Hi @arunxlnx 

Thank you for Alexnet tip. I have run it again and now I have better performance:

| Network | 6-bit | 8-bit |
|   ------     |  ----- | ------ |
| Alexnet |  8.86 | 8.75  |

About VGG-SSD300 `[INFOx] Design will not fit for input dimension  256x1x1 with filter dimension 3x3` error, using an old *libxstack.so* allows me to run the executable but with a new one no.  

I think that this starts happening after [this commit](2ac2f2fc76e99ec3bdea7f2ccac926c196962cc1), is this correct?

anithabarri, 2018-08-28:
Hi @jlamperez

For the following issue:
**[INFOx] Design will not fit for input dimension  256x1x1 with filter dimension 3x3**

Use the updated init module.

Regards
Anitha

anithabarri, 2018-08-31:
Closing this issue

closed 2018-08-31 by anithabarri

jlamperez, 2018-09-03:
Hi,

I am trying to get the 606.46 fps using Alexnet w/o fcn but I only achive 310. 

[Here](https://github.com/Xilinx/CHaiDNN/files/2346290/alexnet_wofc.cpp.txt) the .cpp application that I have created. In this file I have changed the `string end_layer`   to  "pool5" and remove ping-pong style. 

Is this procedure correct to get 606 performance value or I am missing something?

Thanks!

JIAWENYANG, 2019-01-21:
@jlamperez can you show me the prototxt and caffemodel of the input about SSD algorithm?