Xilinx/CHaiDNN #55: Fail to run tinySSD on CHaiDNN v2

Harris007, 2018-08-23:
Hi,
I tried to run the final model "_iter_290000.caffemodel" of  TinySSD [https://github.com/lampsonSong/tinySSD](url) on  CHaiDNN v2. 
I successfully generate quantization prototxt using XportDNN in both two quantization mode but run failure. Both quantization prototxt come to the same error: " ["EO005"] Layer opcode: Layer is not support : BatchNorm"
error log attached below:
[log.txt](https://github.com/Xilinx/CHaiDNN/files/2314583/log.txt)

I checked all the layers of tinySSD. All the layer types are supported  [https://github.com/Xilinx/CHaiDNN/blob/master/docs/SUPPORTED_LAYERS.md](url)

Can anyone help me to run?

Thanks a lot, 

Harris

Harris007, 2018-09-07:
do you find any solution to help me run this model? @smenon009 @koppada-raju

smenon009, 2018-09-07:
Hi @Harris007 
At the moment, TinySSD is not verified/ tested on CHai. Will update when we do. 

In the meantime, you could try manually fusing the CONV+BN+SCALE layer weights and the BN+SCALE+CONV layer weights to CONV layers with the weights updated.  So, Manual Fusion(Model & prototxt) --> XportDNN --> CHai will be the sequence of steps.

KFT-JunYS, 2018-09-14:
@smenon009 
[tinySSD.zip](https://github.com/Xilinx/CHaiDNN/files/2381485/tinySSD.zip)
I tried to do your comments but it has a bad result. I attached model and weights files of tinySSD.
As you mentioned, this files has not batch norm and scale layer.

Harris007, 2018-11-07:
hi @smenon009 
Thanks for your suggestion. I have fused bn,scale,conv weights into conv weights.  And  the fused model achieves same precision with the same test dataset(2700 images).  I used XportDNN to produce prototxts, both Xilinx-Quantization and DynamicFixed using default parameters. The fused model successfully runing in ZCU102 but produce very bad result, no predictive box for DynamicFixed prototxt and and many small predicive boxes for Xilinx-Quantization prototxt. I think there is something wrong when using  XportDNN quantize the model.
Could you help me solve this problem? It is only one step to running tinyssd in the platform.

regards,

Harris