Xilinx/CHaiDNN #56: Fail to build CHaiDNN on ZCU104 platform on Xilinx SDSOC 2018.2

tuananhcoder, 2018-08-24:
Hello,
I got issue when I was trying to build CHaiDNN version 2 on ZCU104 platform on SDSoC 2018.2.
`===>The following messages were generated while processing /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/vivado/prj/prj.runs/impl_1 :
ERROR: [VPL 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 354597 of such cell types but only 230400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [VPL 30-640] Place Check : This design requires more LUT as Memory cells than are available in the target device. This design requires 156785 of such cell types but only 101760 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [VPL 30-640] Place Check : This design requires more LUT as Distributed RAM cells than are available in the target device. This design requires 152686 of such cell types but only 101760 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [VPL 30-640] Place Check : This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 433 of such cell types but only 312 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 1102 of such cell types but only 624 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-640] Place Check : This design requires more RAMB36E2 cells than are available in the target device. This design requires 433 of such cell types but only 312 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-640] Place Check : This design requires more RAMD64E cells than are available in the target device. This design requires 113072 of such cell types but only 101760 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-99] Placer failed with error: 'Implementation Feasibility check failed, Please see the previously displayed individual error or warning messages for more details.'
Please review all ERROR and WARNING messages during placement to understand the cause for failure.
ERROR: [VPL 17-69] Command failed: Placer could not place all instances
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, place_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/bin/vpl   --iprepo /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/iprepo/repo  --iprepo /mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/data/ip/xilinx  --platform /mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/platforms/zcu104/zcu104.xpfm  --temp_dir /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0  --output_dir /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/vpl  --input_file /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:XiDeconvTop:XiConvolutionTop:adapter --webtalk_flag SDSoC  --remote_ip_cache /home/anhnt/workspace/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/anhnt/workspace/zcu104_CHaiDNNv2/src/design/conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/anhnt/workspace/zcu104_CHaiDNNv2/src/design/conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: recipe for target 'zcu104_CHaiDNNv2.elf' failed
make: *** [zcu104_CHaiDNNv2.elf] Error 1`

So you guys successd build CHaiDNN v2 on ZCU104 can share to me?
Thank you
T.Anh.

nkpavan, 2018-08-24:
Hi @tuananhcoder ,

You are seeing this error as all the storage structures in the design are mapped to BRAMs. zcu104 has URAMs and some storage structures need to be mapped to URAMs to fit the design on zcu104. This can be done by setting the below macros in [xi_conv_config.h](https://github.com/Xilinx/CHaiDNN/blob/master/design/conv/include/xi_conv_config.h).

#define XI_WTS_URAM_EN 	1
#define XI_ISTG_URAM_EN 	1
#define XI_OSTG_URAM_EN 	1

After setting the above macros, run make POOL_ENABLE=0 DECONV_ENABLE=0 to initiate the build.

tuananhcoder, 2018-08-24:
Hi @nkpavan ,
Thank you so much,
I am building the project.

tuananhcoder, 2018-08-27:
Hi @nkpavan ,
I still met error after I set like you guide:
`===>The following messages were generated while processing /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/vivado/prj/prj.runs/impl_1 :
ERROR: [VPL 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 677 of such cell types but only 624 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-99] Placer failed with error: 'IO Clock Placer stopped due to earlier errors. Implementation Feasibility check failed, Please see the previously displayed individual error or warning messages for more details.'
Please review all ERROR and WARNING messages during placement to understand the cause for failure.
ERROR: [VPL 17-69] Command failed: Placer could not place all instances
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, place_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/bin/vpl   --iprepo /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/iprepo/repo  --iprepo /mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/data/ip/xilinx  --platform /mnt/F888112AA89C0E54/ProgramFile/Xilinx/SDx/2018.2/platforms/zcu104/zcu104.xpfm  --temp_dir /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0  --output_dir /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/vpl  --input_file /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:XiDeconvTop:XiConvolutionTop:adapter --webtalk_flag SDSoC  --remote_ip_cache /home/anhnt/workspace/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/anhnt/workspace/zcu104_CHaiDNNv2/src/design/conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/anhnt/workspace/zcu104_CHaiDNNv2/src/design/conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/anhnt/workspace/zcu104_CHaiDNNv2/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:60: recipe for target 'zcu104_CHaiDNNv2.elf' failed
make: *** [zcu104_CHaiDNNv2.elf] Error 1`

After met error, I try to build with setting the below more but still met same error.
#define XI_BIAS_URAM_EN 	1
#define XI_FEED_URAM_EN 	1
#define XI_SCALE_URAM_EN 	1

nkpavan, 2018-08-27:
Hi @tuananhcoder ,

Please share sds.rpt, sds.log and sds_xi_convolution_top.rpt files present in _sds/reports folder with us to debug the issue.

tuananhcoder, 2018-08-27:
Hi @nkpavan ,
[reports.zip](https://github.com/Xilinx/CHaiDNN/files/2322720/reports.zip)
Here are two file sds.log and sds_xi_convolution_top.rpt,
I didn't find file sds.rpt.
Thanks.

nkpavan, 2018-08-27:
I went through the reports and observed URAM over utilization. You mentioned in your previous comment that you enabled below macros as well. 
#define XI_BIAS_URAM_EN 1
#define XI_FEED_URAM_EN 1
#define XI_SCALE_URAM_EN 1
Are reports that you sent correspond to this configuration? 

Please send your xi_conv_config.h file as well.

tuananhcoder, 2018-08-28:
Here is my xi_conv_config.h file.
[xi_conv_config.h.zip](https://github.com/Xilinx/CHaiDNN/files/2326218/xi_conv_config.h.zip)

nkpavan, 2018-08-29:
Please use this [xi_conv_config.zip](https://github.com/Xilinx/CHaiDNN/files/2331474/xi_conv_config.zip) file and build the design. This will build a design with 512 compute DSPs. Also make sure in the make command that POOL_ENABLE and DECONV_ENABLE are set to 0.

salcanmor, 2018-09-17:
@nkpavan To compile CHaiDNN in ZCU104, did you change #define XI_PIX_PROC **32** by #define XI_PIX_PROC **16**  in hw_settings.h ?

nkpavan, 2018-09-17:
@salcanmor ,

To build a design for zcu104 with 512 compute DSPs, both XI_KER_PROC and XI_PIX_PROC should be set to 16. Also the settings in the xi_conv_config.h file and hw_settings.h file should be same. xi_conv_config.h file is used for hardware build and hw_settings.h file is used for software build.

closed 2018-10-11 by nkpavan

Okaymaddy, 2020-06-05:
@salcanmor @nkpavan @VishalX @tuananhcoder I am trying to build ChaiDNN for zcu102 using SDx 2018.2 in Ubuntu 18.04. I keep getting the following error:

`ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, place_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/saikat/Desktop/SDx/2018.2/bin/vpl   --iprepo /home/saikat/CHaiDNN/design/build/_sds/iprepo/repo  --iprepo /home/saikat/Desktop/SDx/2018.2/data/ip/xilinx  --platform /home/saikat/Desktop/SDx/2018.2/platforms/zcu102/zcu102.xpfm  --temp_dir /home/saikat/CHaiDNN/design/build/_sds/p0  --output_dir /home/saikat/CHaiDNN/design/build/_sds/p0/vpl  --input_file /home/saikat/CHaiDNN/design/build/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels XiConvolutionTop:PoolTop:XiDeconvTop:adapter --webtalk_flag SDSoC  --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/saikat/CHaiDNN/design/build/../conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/saikat/CHaiDNN/design/build/../conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/saikat/CHaiDNN/design/build/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:147: recipe for target 'libxlnxdnn.so' failed
make: *** [libxlnxdnn.so] Error 1`

Any help or suggestions on how to fix this?