Xilinx/CHaiDNN #57: Cant compile the Latest version.

adu81020799, 2018-08-27:
I tried compiling the latest version of Chai DNN on zcu102 platform by following the steps as discussed using following guide:
https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md
but when i try to build it i get the following error.
Create data motion intermediate representation
clang: error: no such file or directory: '/home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/Release/__DSP48E2__'
clang: error: no such file or directory: '/home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/Release/__SYNTHESIS__'
ERROR: [SdsCompiler 83-5005] clang exited with non-zero code processing /home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/Release/__DSP48E2__ /home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/Release/__SYNTHESIS__ /home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/src/software/xtract/caffe.pb.cc
sds++ log file saved as /home/uncc_embed/workspace/sdx/CHAI/CHaiDNN/Release/_sds/reports/sds_caffe.pb.log
ERROR: [SdsCompiler 83-5004] Build failed

src/software/xtract/subdir.mk:37: recipe for target 'src/software/xtract/caffe.pb.o' failed
make: *** [src/software/xtract/caffe.pb.o] Error 1

14:53:57 Build Finished (took 5s.643ms)

I tried adding the folders by myself but it still gives the error. 

Please help.

xiaokaijia1988, 2018-08-29:
I have the same question,but i don't know to solve it

tuananhcoder, 2018-08-30:
May be your Xilinx SDx has problem. Where did you install Xilinx SDx? What version is it?

adu81020799, 2018-08-30:
The Version is 2018.2 , I am using Linux  16.04

bennihoffmann, 2018-09-19:
Hi,
It is a copy and past error. Probably you were following the 'Creating SDx GUI Project' guide like I did when the problem occurred. 

**I noticed that, pasting the symbols in SDx an space was inserted in front of each Symbol (e.g. '  __DSP48E2__' instead of ' __DSP48E2__') which was hard to recognized from the GUI. I could fixed it by type in all symbols manually.**

Hope it will work for you, Ben