Xilinx/CHaiDNN #60: Build failure

adu81020799, 2018-09-10:
I was trying to build the Chain DNN using Sdx tool. Currently I am getting  two prominent errors:
1.unknown type name '__Float16x4_t'	CHaiDNN		line 42, external location: /opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h	C/C++ Problem
2.unknown type name '__Int32x2_t'; did you mean '__int32_t'?	CHaiDNN		line 40, external location: /opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h	C/C++ Problem
 All the  other error looks similar to above and are related to the arm_neon.h

Please let me know if I am missing something

adu81020799, 2018-09-10:
I am using SDx 2018.2

salcanmor, 2018-09-11:
same problem here, any help @VishalX ?

VishalX, 2018-10-08:
@adu81020799 @salcanmor,

Please make the below change and try the build.
In [xi_eltwiseadd_top.cpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/swkernels/xi_eltwiseadd_top.cpp) file, line 38,
instead of `#ifdef`, use `#ifndef`.

We will update the code to fix the issue soon.

adu81020799, 2018-10-08:
Hello Vishal,
I tried to compile it with the said settings. I got new errors:

Generating data motion network
ERROR: Linking globals named 'clock_start': symbol multiply defined!
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2018.2/bin/llvm-link -o /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/sds_all.o /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/caffe.pb.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/caffe_network_parser.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/xgraph.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/xi_funcs.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/xtract_opt.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/xtract/xtract_utility.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_crop.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_eltwiseadd_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_fc_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_format_converter_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_nms_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_norm_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_permute_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/swkernels/xi_swsoftmax_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/scheduler/xi_perf_eval.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/scheduler/xi_scheduler.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/scheduler/xi_thread_routines.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/scheduler/xi_utils.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/interface/xi_interface.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/interface/xi_readwrite_util.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/init/xi_init.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/include/xchange_structs.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/imageread/xi_input_image.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/alexnet_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/alexnetfcn_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/eval_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/googlenet_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/resnet50_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/ssd_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/sw_layer_plugin.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/example/vgg_ex.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/custom/custom_class.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/common/kernelinfo_class.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/common/xi_kernels.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/checkers/checkers.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/software/bufmgmt/xi_buf_mgmt.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/design/wrapper/dnn_wrapper.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/design/pool/src/pooling_layer_dp_2xio_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/design/deconv/src/xi_deconv_top.s /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/.llvm/./src/design/conv/src/xi_convolution_top.s'
sds++ log file saved as /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [CHaiDNN.elf] Error 1
makefile:60: recipe for target 'CHaiDNN.elf' failed

VishalX, 2018-10-08:
@adu81020799 ,

That's because you have added multiple example files in your project. Please only use one example file and remove others from the build.

adu81020799, 2018-10-09:
hello Vishal,
I have tried it . Sorry for not updating on the forums. I tried building it by keeping a single example. I t did resolve the previous error but while synthesizing  I got following error.
Creating Vivado project and starting FPGA synthesis.

===>The following messages were generated while  Compiling (top level synthesis checkpoint) dynamic region Log file: /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/p0/vivado/prj/prj.runs/synth_1/runme.log  :
ERROR: [VPL-1] couldn't read file "/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/conv/scripts/mcps.tcl": no such file or directory
ERROR: [VPL 60-704] Integration error, One or more synthesis runs failed during dynamic region dcp generation
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2018.2/bin/vpl   --iprepo /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/iprepo/repo  --iprepo /opt/Xilinx/SDx/2018.2/data/ip/xilinx  --platform /opt/Xilinx/SDx/2018.2/platforms/zcu102/zcu102.xpfm  --temp_dir /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/p0  --output_dir /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/p0/vpl  --input_file /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:XiDeconvTop:XiConvolutionTop:adapter --webtalk_flag SDSoC  --remote_ip_cache /home/uncc_embed/workspace/MLDNN/ip_cache --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/uncc_embed/workspace/MLDNN/CHaiDNN/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed
make: *** [CHaiDNN.elf] Error 1

makefile:60: recipe for target 'CHaiDNN.elf' failed

This is showing up because the make file which gets generated for the release shows the path  tries to use the path   {/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/conv/scripts/mcps.tcl  but the mcps.tcl is actually present in {/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/design/conv/scripts/mcps.tcl . Can you tell me whic file i need to update in order to change this path ?

VishalX, 2018-10-09:
Does the below path exists?
`/home/uncc_embed/workspace/MLDNN/CHaiDNN/src/conv/scripts/mcps.tcl`.

If not, See the step 20 in [SDx-GUI](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md) guide and correct the path of `mcps.tcl` file in SDS++ Linker Settings.

salcanmor, 2018-10-10:
@VishalX ,
 instead of #ifdef, use #ifndef <--- that did not work for me.

I'm still getting the same error:

(c) Copyright 2012-2018 Xilinx, Inc. All Rights Reserved.
#-----------------------------------------------------------
# Tool version  : sds++ 2018.2 SW Build 2258646 on Thu Jun 14 20:03:40 MDT 2018
# Start time    : Wed Oct 10 16:21:59 CEST 2018
# Command line  : sds++ -sds-hw XiConvolutionTop /home/pc-i7/workspace/CHaiDNN/src/design/conv/src/xi_convolution_top.cpp -clkid 1 -hls-tcl /home/pc-i7/workspace/CHaiDNN/src/design/conv/scripts/config_core.tcl -sds-end -D__SDSOC -D__CONV_ENABLE__ -D__DSP48E2__ -D__SYNTHESIS__ -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -Wall -O0 -I../src -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/opencv/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/protobuf/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/cblas/arm64/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/src/ -c -fmessage-length=0 -MTsrc/software/example/ssd_ex.o -MMD -MP -MFsrc/software/example/ssd_ex.d -MTsrc/software/example/ssd_ex.o -o src/software/example/ssd_ex.o ../src/software/example/ssd_ex.cpp -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf zcu104
# Log file      : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_ssd_ex.log
# Journal file  : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_ssd_ex.jou
# Report file   : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_ssd_ex.rpt
#-----------------------------------------------------------

Create data motion intermediate representation
/opt/Xilinx/SDx/2018.2/bin/clang_wrapper -I../src -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/opencv/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/protobuf/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/cblas/arm64/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/src/ -D__SDSOC -D__CONV_ENABLE__ -D__DSP48E2__ -D__SYNTHESIS__ -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -Wall -fmessage-length=0 -MMD -MP -D __SDSCC__ -target aarch64-linux-gnu -mcpu=cortex-a53 -O0 -g     -I /opt/Xilinx/SDx/2018.2/target/aarch64-linux/include  -D HLS_NO_XIL_FPO_LIB  -I /opt/Xilinx/Vivado/2018.2/include   -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1 -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/aarch64-linux-gnu -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/backward -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include-fixed -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include  -std=c++11 -emit-llvm -S /home/pc-i7/workspace/CHaiDNN/src/software/example/ssd_ex.cpp -o /home/pc-i7/workspace/CHaiDNN/Release/_sds/.llvm/src/software/example/ssd_ex.s
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:30:13: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma GCC push_options
            ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:31:13: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma GCC target ("+nothing+simd")
            ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:38:9: error: unknown type name '__Int8x8_t'
typedef __Int8x8_t int8x8_t;
        ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:39:9: error: unknown type name '__Int16x4_t'
typedef __Int16x4_t int16x4_t;
        ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:40:9: error: unknown type name '__Int32x2_t'; did you mean '__int32_t'?
typedef __Int32x2_t int32x2_t;
        ^~~~~~~~~~~
        __int32_t
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:40:20: note: '__int32_t' declared here
typedef signed int __int32_t;
                   ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:41:9: error: unknown type name '__Int64x1_t'; did you mean '__int64_t'?
typedef __Int64x1_t int64x1_t;
        ^~~~~~~~~~~
        __int64_t
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:43:25: note: '__int64_t' declared here
typedef signed long int __int64_t;
                        ^
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:42:9: error: unknown type name '__Float16x4_t'
typedef __Float16x4_t float16x4_t;
        ^
[... cut ...]
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include/arm_neon.h:56:9: error: unknown type name '__Float32x4_t'
typedef __Float32x4_t float32x4_t;
        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
ERROR: [SdsCompiler 83-5005] clang exited with non-zero code processing /home/pc-i7/workspace/CHaiDNN/src/software/example/ssd_ex.cpp
sds++ log file saved as /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_ssd_ex.log
ERROR: [SdsCompiler 83-5004] Build failed
sds++ completed at Wed Oct 10 16:22:01 CEST 2018

xiaokaijia1988, 2018-10-11:
@VishalX my error

/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:1600:5:   required from ‘ap_private<_AP_W, _AP_S, true>::ap_private(long unsigned int) [with int _AP_W = 40; bool _AP_S = true]’
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:1943:50:   required from ‘typename ap_private<_AP_W, _AP_S, true>::RType<_AP_W2, _AP_S2>::mult ap_private<_AP_W, _AP_S, true>::operator*(const ap_private<_AP_W1, _AP_S1>&) const [with int _AP_W1 = 16; bool _AP_S1 = false; int _AP_W = 24; bool _AP_S = true; typename ap_private<_AP_W, _AP_S, true>::RType<_AP_W2, _AP_S2>::mult = ap_private<40, true, true>]’
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp:6052:60:   required from here
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:2057:98: warning: enum constant in boolean context [-Wint-in-bool-context]
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h: In instantiation of ‘void ap_private<_AP_W, _AP_S, true>::clearUnusedBits() [with int _AP_W = 41; bool _AP_S = true]’:
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:1600:5:   required from ‘ap_private<_AP_W, _AP_S, true>::ap_private(long unsigned int) [with int _AP_W = 41; bool _AP_S = true]’
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:1990:51:   required from ‘typename ap_private<_AP_W, _AP_S, true>::RType<_AP_W2, _AP_S2>::plus ap_private<_AP_W, _AP_S, true>::operator+(const ap_private<_AP_W1, _AP_S1>&) const [with int _AP_W1 = 24; bool _AP_S1 = true; int _AP_W = 40; bool _AP_S = true; typename ap_private<_AP_W, _AP_S, true>::RType<_AP_W2, _AP_S2>::plus = ap_private<41, true, true>]’
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp:6052:79:   required from here
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h:2057:98: warning: enum constant in boolean context [-Wint-in-bool-context]
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp: In instantiation of ‘void ProcIStagingBuff_fb(input_struct, conv_struct, output_struct, weight_struct, gmem_inputtype_layer1*, gmem_inputtype_layerx*, gmem_inputtype_layerx*, gmem_outputtype*, gmem_outputtype*, gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_biastype*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, bool) [with int IN_WW = 227; int IN_HH = 227; int OUT_WW = 55; int OUT_HH = 55; int CNUM_KERNELS = 96; int CFILTER_SIZE = 11; int CCONV_STRIDE = 4; int PPOOL_STRIDE = 2; int PPOOL_SIZE = 3; int IINPUT_PLANES = 4; int PNKPF = 8; input_struct = input_descriptor; conv_struct = conv_descriptor; output_struct = output_descriptor; weight_struct = weight_descriptor; gmem_inputtype_layer1 = long long unsigned int; gmem_inputtype_layerx = ap_uint<128>; gmem_outputtype = ap_uint<128>; gmem_weighttype = ap_uint<128>; gmem_biastype = long long unsigned int]’:
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp:7794:2:   required from ‘void ConvLayer_fa(input_struct, weight_struct, output_struct, conv_struct, group_conv_struct, gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_inputtype_layer1*, gmem_inputtype_layerx*, gmem_inputtype_layerx*, gmem_biastype*, gmem_outputtype*, gmem_outputtype*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, bool) [with int IN_WW = 227; int IN_HH = 227; int OUT_WW = 55; int OUT_HH = 55; int CNUM_KERNELS = 96; int CFILTER_SIZE = 11; int CCONV_STRIDE = 4; int PPOOL_STRIDE = 2; int PPOOL_SIZE = 3; int IINPUT_PLANES = 4; int PNKPF = 8; input_struct = input_descriptor; weight_struct = weight_descriptor; output_struct = output_descriptor; conv_struct = conv_descriptor; group_conv_struct = group_conv_descriptor; gmem_weighttype = ap_uint<128>; gmem_inputtype_layer1 = long long unsigned int; gmem_inputtype_layerx = ap_uint<128>; gmem_biastype = long long unsigned int; gmem_outputtype = ap_uint<128>]’
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp:7854:63:   required from ‘void Convolution(gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_weighttype*, gmem_outputtype*, gmem_outputtype*, gmem_inputtype_layerx*, gmem_inputtype_layerx*, gmem_inputtype_layer1*, gmem_biastype*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, gmem_inputtype_layer1*, int*, bool) [with int IN_WW = 227; int IN_HH = 227; int OUT_WW = 55; int OUT_HH = 55; int CNUM_KERNELS = 96; int CFILTER_SIZE = 11; int CCONV_STRIDE = 4; int PPOOL_STRIDE = 2; int PPOOL_SIZE = 3; int IINPUT_PLANES = 4; int PNKPF = 8; gmem_weighttype = ap_uint<128>; gmem_outputtype = ap_uint<128>; gmem_inputtype_layerx = ap_uint<128>; gmem_inputtype_layer1 = long long unsigned int; gmem_biastype = long long unsigned int]’
xi_convolution_top.cpp:123:32:   required from here
/home/hrxkj/workspace/CHaiDNN/src/design/conv/src/xi_convolution.hpp:7475:2: warning: label ‘ProciStg_Loop’ defined but not used [-Wunused-label]
  ProciStg_Loop:
  ^~~~~~~~~~~~~
                 from /home/Xilinx/Vivado/2018.2/include/ap_int.h:65,
                 from /home/hrxkj/workspace/CHaiDNN/src/design/conv/include/xi_conv_config.h:20,
                 from xi_convolution_top.cpp:17:
/home/Xilinx/Vivado/2018.2/include/etc/ap_private.h: In instantiation of ‘void ap_private<_AP_W, _AP_S, true>::clearUnusedBits() [with int _AP_W = 7; bool _AP_S = true]’:
[... cut ...] "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/hrxkj/workspace/CHaiDNN/src/design/conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/hrxkj/workspace/CHaiDNN/src/design/conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/hrxkj/workspace/CHaiDNN/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [CHaiDNN.elf] 错误 1

16:29:42 Build Finished (took 19m:55s.21ms)

salcanmor, 2018-10-15:
@VishalX ,

`#ifndef` disables the block under it, from line 38 to line 398.
But the error related to arm_neon.h is still there. Any idea on how to solve it ?

VishalX, 2018-10-15:
@salcanmor,

Can you check which from which source file is causing this issue? Please check the log and let me know.

salcanmor, 2018-10-15:
@VishalX ,

I solved that error. But another one appeared, any idea on how to solve that?
```
(c) Copyright 2012-2018 Xilinx, Inc. All Rights Reserved.
#-----------------------------------------------------------
# Tool version  : sds++ 2018.2 SW Build 2258646 on Thu Jun 14 20:03:40 MDT 2018
# Start time    : Mon Oct 15 11:02:58 CEST 2018
# Command line  : sds++ -sds-hw XiConvolutionTop /home/pc-i7/workspace/CHaiDNN/src/design/conv/src/xi_convolution_top.cpp -clkid 1 -hls-tcl /home/pc-i7/workspace/CHaiDNN/src/design/conv/scripts/config_core.tcl -sds-end -D__SDSOC -D__CONV_ENABLE__ -D__DSP48E2__ -D__SYNTHESIS__ -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -Wall -O0 -I../src -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/opencv/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/protobuf/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/cblas/arm64/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/src/ -c -fmessage-length=0 -MTsrc/design/build/_sds/swstubs/xi_convolution_top.o -MMD -MP -MFsrc/design/build/_sds/swstubs/xi_convolution_top.d -MTsrc/design/build/_sds/swstubs/xi_convolution_top.o -o src/design/build/_sds/swstubs/xi_convolution_top.o ../src/design/build/_sds/swstubs/xi_convolution_top.cpp -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf zcu104
# Log file      : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_xi_convolution_top.log
# Journal file  : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_xi_convolution_top.jou
# Report file   : /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_xi_convolution_top.rpt
#-----------------------------------------------------------

Create data motion intermediate representation
/opt/Xilinx/SDx/2018.2/bin/clang_wrapper -I../src -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/opencv/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/protobuf/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/cblas/arm64/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/src/ -D__SDSOC -D__CONV_ENABLE__ -D__DSP48E2__ -D__SYNTHESIS__ -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -Wall -fmessage-length=0 -MMD -MP -D __SDSCC__ -target aarch64-linux-gnu -mcpu=cortex-a53 -O0 -g     -I /opt/Xilinx/SDx/2018.2/target/aarch64-linux/include  -D HLS_NO_XIL_FPO_LIB  -I /opt/Xilinx/Vivado/2018.2/include   -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1 -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/aarch64-linux-gnu -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/backward -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include-fixed -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include  -std=c++11 -emit-llvm -S /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/swstubs/xi_convolution_top.cpp -o /home/pc-i7/workspace/CHaiDNN/Release/_sds/.llvm/src/design/build/_sds/swstubs/xi_convolution_top.s
Performing accelerator source linting for XiConvolutionTop
/opt/Xilinx/SDx/2018.2/bin/sdslint -target cortex-a53 -func "XiConvolutionTop" /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/swstubs/xi_convolution_top.cpp -- -c -I../src -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/opencv/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/protobuf/arm64/include/ -I/home/pc-i7/Descargas/CHaiDNN-master/SD_Card/cblas/arm64/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/conv/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/pool/src/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/include/ -I/home/pc-i7/workspace/CHaiDNN/src/design/deconv/src/ -D__SDSOC -D__CONV_ENABLE__ -D__DSP48E2__ -D__SYNTHESIS__ -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -Wall -fmessage-length=0 -MMD -MP -D __SDSCC__ -target aarch64-linux-gnu -mcpu=cortex-a53 -O0 -g -w    -I /opt/Xilinx/SDx/2018.2/target/aarch64-linux/include  -D HLS_NO_XIL_FPO_LIB  -I /opt/Xilinx/Vivado/2018.2/include   -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1 -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/aarch64-linux-gnu -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/backward -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include-fixed -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/include -I/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/aarch64-l [... cut ...]
"hls_proc $argv"
INFO: [Common 17-206] Exiting vivado_hls at Mon Oct 15 11:03:05 2018...
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/Vivado/2018.2/bin/vivado_hls /home/pc-i7/workspace/CHaiDNN/Release/_sds/vhls/XiConvolutionTop_run.tcl -l XiConvolutionTop_vivado_hls.log'
sds++ log file saved as /home/pc-i7/workspace/CHaiDNN/Release/_sds/reports/sds_xi_convolution_top.log
ERROR: [SdsCompiler 83-5004] Build failed
sds++ completed at Mon Oct 15 11:03:05 CEST 2018
```

VishalX, 2018-10-15:
`ERROR: [SYNCHK 200-79] Cannot find the top function '' in the design. Possible causes are: (1) the top function name is misspelled; (2) the top function is nonexistent or declared as static.`

The top function name is missing. Please check the settings.

VishalX, 2018-10-19:
@salcanmor,

`Top` functions are the functions you're marking as hardware accelerators, not the `main` function.
See [Comment](https://github.com/Xilinx/CHaiDNN/issues/74#issuecomment-431300944)

salcanmor, 2018-10-22:
@VishalX 

I'm experiencing something weird. If I write `sds++ -sds-hw XiConvolutionTop /home/...` in SDS++ Compiler (like in in step 16), what I get is 

> ERROR: [SYNCHK 200-79] Cannot find the top function '' in the design. 

But, If I write any other thing, for example, the same function in lowercase, what I get is:

> ERROR: [SYNCHK 200-79] Cannot find the top function 'xiconvolutiontop' in the design.

So, the problem comes what I write `XiConvolutionTop`, which has no sense for me.

Can you help me to solve this problem please?

salcanmor, 2018-10-22:
@VishalX 

I'm experiencing something weird. If I write `sds++ -sds-hw XiConvolutionTop /home/...` in SDS++ Compiler (like in in step 16), what I get is 

> ERROR: [SYNCHK 200-79] Cannot find the top function '' in the design. 

But, If I write any other thing, for example, the same function in lowercase, what I get is:

> ERROR: [SYNCHK 200-79] Cannot find the top function 'xiconvolutiontop' in the design.

So, the problem comes what I write `XiConvolutionTop`, which has no sense for me.

Can you help me to solve this problem please? It seems like something is hardcoded

VishalX, 2018-10-22:
@salcanmor,

I'm not sure why are you are seeing these issues. I've successfully rebuilt the design following the [BUILD_USING_SDX_GUI.md](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md) document. 

Please try creating a fresh workspace with a new project and try again.

VishalX, 2018-10-23:
Closed #83 #82 #81 
@salcanmor,
Please keep only one thread open for an specific issue.

salcanmor, 2018-10-23:
@VishalX 

I managed to build CHaiDNN successfully, but I had to remove the symbols __POOL_ENABLE__ and __DECONV_ENABLE__.

VishalX, 2018-10-23:
@salcanmor,

In this case, the Deconv accelerator will not be present in the build and it will use the integrated pool.
To build all the accelerators, you have to enable those symbols.

Can you also try with both symbols again?
Make sure, there are no whitespaces introduced when you add the symbols in build settings.

salcanmor, 2018-10-23:
@VishalX 

Does pool and deconv fit in ZCU104 ? I think the problem is that, what do you think ?

nkpavan, 2018-10-23:
@salcanmor ,

For ZCU104 with 512 DSP config, Deconv and Pool will not fit. The pooling functionality implemented in the Convolution kernel is used in this case. Please close the issue if your issue is resolved.

closed 2018-10-29 by VishalX