Xilinx/CHaiDNN #63: ERROR: [VPL RTSTAT-6] Partial route conflicts

salcanmor, 2018-09-13:
Hi guys,

I'm trying to "Build from Source" the hardware using make ...

Does anyone know why I'm getting this error message?

I'm targeting ZCU104 with this options:

**xi_conv_config.h:**

//**** URAM ENABLE FLAGS FOR THE BUFFERS
#define XI_BIAS_URAM_EN 	1
#define XI_WTS_URAM_EN 		1
#define XI_ISTG_URAM_EN 	1
#define XI_OSTG_URAM_EN 	1
#define XI_FEED_URAM_EN 	1
#define XI_SCALE_URAM_EN 	1

#define XI_KER_PROC       	16
#define XI_PIX_PROC       	16

**hw_settings.h:**

#define XI_KER_PROC       	16
#define XI_PIX_PROC       	16

**ERROR MESSAGES:**

===>The following messages were generated while  Compiling (bitstream) accelerator binary: bin Log file: /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/p0/vivado/prj/prj.runs/impl_1/runme.log  :
ERROR: [VPL RTSTAT-6] Partial route conflicts: 12263 net(s) have a partial conflict. The problem bus(es) and/or net(s) are zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRARDADDR[0], zcu104_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_Conv3dWtsRead_fu_2241/ADDRARDADDR[1], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_readData_16_2048_s_fu_1111/ADDRARDADDR[2], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRARDADDR[2], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_readData_16_2048_s_fu_1111/ADDRARDADDR[4], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRARDADDR[4], zcu104_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_Conv3dWtsRead_fu_2241/ADDRARDADDR[6], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_readData_16_2048_s_fu_1111/ADDRARDADDR[9], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_readData_16_2048_s_fu_1111/ADDRARDADDR[10], zcu104_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1803/grp_ProcFeedingBuff_fz_fu_1086/grp_ProcInputBuff_fc_fu_711/grp_ProcWeightBuff_fd_fu_1213/grp_ProcResult_fe_fu_686/result_pong_fe0_3_V_U/a2_ProcResult_fe_resVhK_ram_U/ADDRBWRADDR[0], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRBWRADDR[0], zcu104_i/XiConvolutionTop_1/inst/grp_Convolution_fu_236/grp_ConvLayer_fa_fu_218/grp_ProcIStagingBuff_fb_fu_1284/grp_ProcFeedingBuff_En_f_fu_1803/grp_ProcFeedingBuff_fz_fu_1086/grp_ProcInputBuff_fc_fu_711/grp_ProcWeightBuff_fd_fu_1213/grp_ProcResult_fe_fu_686/result_ping_fe0_11_s_U/a2_ProcResult_fe_resVhK_ram_U/ADDRBWRADDR[3], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRBWRADDR[5], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRBWRADDR[8], zcu104_i/PoolTop_1/inst/grp_poolKernel_fu_128/grp_PlanesPingPong_fu_397/grp_poolingLayer_fu_974/ADDRBWRADDR[9]... and (the first 15 of 12178 listed), GLOBAL_LOGIC0.
ERROR: [VPL 12-1345] Error(s) found during DRC. Bitgen not run.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, write_bitstream ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2018.2/bin/vpl   --iprepo /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/iprepo/repo  --iprepo /opt/Xilinx/SDx/2018.2/data/ip/xilinx  --platform /opt/Xilinx/SDx/2018.2/platforms/double_clk_platforms/zcu104/zcu104.xpfm  --temp_dir /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/p0  --output_dir /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/p0/vpl  --input_file /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels XiConvolutionTop:PoolTop:XiDeconvTop:adapter --webtalk_flag SDSoC  --xp "param:compiler.skipTimingCheckAndFrequencyScaling=1" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={/home/pc-i7/workspace/CHaiDNN/src/design/build/../conv/scripts/mcps.tcl}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={/home/pc-i7/workspace/CHaiDNN/src/design/build/../conv/scripts/mcps.tcl}" --xp "param:compiler.deleteDefaultReportConfigs=false" '
sds++ log file saved as /home/pc-i7/workspace/CHaiDNN/src/design/build/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:152: recipe for target 'libxlnxdnn.so' failed
make: *** [libxlnxdnn.so] Error 1

nkpavan, 2018-09-17:
@salcanmor ,

As mentioned in #56 , for zcu104, enabling all the macros to map storage structures in the design to URAMs will lead to URAM over-utilization. So enable only the below macros and try building.

#define XI_WTS_URAM_EN    1
#define XI_ISTG_URAM_EN 	1
#define XI_OSTG_URAM_EN 	1

salcanmor, 2018-09-17:
Hi @nkpavan,

I have solved this issue. Thank you.
Here are my settings (for ZCU104):

**xi_conv_config.h:**

//**** URAM ENABLE FLAGS FOR THE BUFFERS
#define XI_BIAS_URAM_EN 	0
#define XI_WTS_URAM_EN 		1
#define XI_ISTG_URAM_EN 	1
#define XI_OSTG_URAM_EN 	1
#define XI_FEED_URAM_EN 	0
#define XI_SCALE_URAM_EN 	0

#define XI_KER_PROC 16
#define XI_PIX_PROC 16

**hw_settings.h:**

#define XI_KER_PROC 16
#define XI_PIX_PROC 16

In **src/design/build/**, you have to make:
make ultraclean 
make POOL_ENABLE=0 DECONV_ENABLE=0

PD. To help other guys with so much troubles when compiling CHaiDNN (like me). I'd like to make a "compilation guide for dummies".
Is there any site where you recommend me to upload that guide?

VishalX, 2018-09-21:
@salcanmor 

You can share your guide on [GitHub**Gist**](https://gist.github.com/ ) and share the link here.
Closing this issue.

closed 2018-09-21 by VishalX

Okaymaddy, 2020-06-04:
@salcanmor Did you write the guide yet?