Xilinx/CHaiDNN #64: Error in make copy

salcanmor, 2018-09-17:
After compiling successfully the hardware (ZCU104), the next step is to run **make copy**, but for some reason, the SD_Card folder has not been created, and I get this error.

root@pc-i7-Z270-HD3P:/home/pc-i7/workspace/CHaiDNN/src/design/build# make copy
cp sd_card/libxlnxdnn.so ../../SD_Card/lib/                                                                                                                                                                                                                                    
cp: cannot create regular file '../../SD_Card/lib/': No such file or directory                                                                                                                                                                                                 
Makefile:156: recipe for target 'copy' failed                                                                                                                                                                                                                                  
make: *** [copy] Error 1                                                                                                                                                                                                                                                       
root@pc-i7-Z270-HD3P:/home/pc-i7/workspace/CHaiDNN/src/design/build#            

How can I solve this error?       
Thanks in advance.

salcanmor, 2018-09-17:
@nkpavan ,

The folder where I cloned the repo and the folder where I have this project (design and software folders) are in different paths, since I tried to compile it first using SDx, and just imported the **design** and **software** folders (following BUILD_USING_SDX_GUI.md).

-Cloned repo: /home/pc-i7/Downloads/CHaiDNN-master/
-Project folder:/home/pc-i7/workspace/CHaiDNN/src/

Should I copy **/home/pc-i7/Downloads/CHaiDNN-master/SD_Card/** to **/home/pc-i7/workspace/CHaiDNN/src/** ?

nkpavan, 2018-09-18:
@salcanmor ,

You can see the commands in the makefile that run when you do make copy, and can manually copy the necessary files to SD_Card folder.  You can refer point 5 in Build CHaiDNN Hardware in README.md file.

closed 2018-10-08 by VishalX