Xilinx/CHaiDNN #69: Read-only file system problem

salcanmor, 2018-09-25:
Hi guys,
I don't know why, when I run a CHaiDNN, it cannot create the outputs of the network. I even can't create a folder.

```
[OUTWR] Write Path : models/GoogleNetWithoutLRN/6Bit/out/out_0_0.txt

[ERROR] can't create file : models/GoogleNetWithoutLRN/6Bit/out/out_0_0.txt

[OUTWR] Write Path : models/GoogleNetWithoutLRN/6Bit/out/out_0_1.txt

[ERROR] can't create file : models/GoogleNetWithoutLRN/6Bit/out/out_0_1.txt
[INFOx] Memory Released
root@zcu104:/mnt# mkdir testdir
mkdir: can't create directory 'testdir': Read-only file system
root@zcu104:/mnt# 
```
How can I disable the **Read-only** issue?

VishalX, 2018-09-28:
@salcanmor ,

Try formatting your SD card and copy the data again. Use FAT32 file system.
If that doesn't work out, try a different card altogether.

Thanks.

VishalX, 2018-10-11:
Closing, Reopen if required.

closed 2018-10-11 by VishalX