Xilinx/CHaiDNN #71: How do I cite CHaiDNN in a research paper?

MohammadSamragh, 2018-09-28:
The paper submission deadline is close so I would appreciate a fast response :) thanks!

AnushaPerla, 2018-09-28:
Hi @MohammadSamragh 

Could you share more on what kind of detail on CHai you want to cite?

MohammadSamragh, 2018-09-28:
Hi @AnushaPerla,

Thank you for the quick reply. I am going to cite it as a framework for FPGA acceleration + compare performance results for Alexnet, 8-bit, ZU9 board.

I would also appreciate if you could let me know whether power consumption measurements are also available for this project.

Many thanks!

AnushaPerla, 2018-10-01:
Hi @MohammadSamragh 

a. In the content, CHaiDNN can be introduced as "A HLS-based Deep Neural Network Accelerator library for Xilinx Ultrascale+ MPSoC devices".
b. And add the GitHub link for CHaiDNN in the references section.

closed 2018-10-11 by AnushaPerla