Xilinx/CHaiDNN #73: [ERROR] can't create file : models/GoogleNetWithoutLRN/6Bit/out/out_0_0.txt

mazixes, 2018-10-03:
Hi,

I am trying to run the inference example on GoogleNet. Here is the last few lines:

```
[INFOx] Init Start : This may take a while ...
[INFOx] Init Done
[INFOx] Image Read done
[INFOx] Input Read Done

[PERFM] Performance with Batching : 141.120655 Images/second

[INFOx] Unapack output data

[OUTWR] Write Path : models/GoogleNetWithoutLRN/6Bit/out/out_0_0.txt

[ERROR] can't create file : models/GoogleNetWithoutLRN/6Bit/out/out_0_0.txt

[OUTWR] Write Path : models/GoogleNetWithoutLRN/6Bit/out/out_0_1.txt

[ERROR] can't create file : models/GoogleNetWithoutLRN/6Bit/out/out_0_1.txt
[INFOx] Memory Released

```
I reformatted the SD card to FAT32 and copied the files again, but the problem persists. The permissions on the 'out' folder is 770.

Thanks.

VishalX, 2018-10-08:
@mazixes ,

Please verify that the path to the .txt files exists.
If the above is fine, try using a new card.

salcanmor, 2018-10-08:
probably, you did not create the folder **out**

mazixes, 2018-10-09:
Thank you both.

I was mistaken because by "models/<network>/out" I thought that I have to create the out folder in "models/GoogleNetWithoutLRN/". But I had to create it in "models/GoogleNetWithoutLRN/6Bit/" which is the full network path.

closed 2018-10-09 by mazixes