Xilinx/CHaiDNN #74: What is the root function?

xkardec, 2018-10-15:
I'm trying to build CHaiDNN. 
In project.sdx, I have chosen "Release".
Next, In "Root function", I choose the main function, but that function does not remain selected.
So, what's the right root function here?
Thank you!

VishalX, 2018-10-16:
@xkardec 

`main` should be the root function. 
Please make sure you're adding only one example file to the project.

xkardec, 2018-10-17:
> @xkardec
> 
> `main` should be the root function.
> Please make sure you're adding only one example file to the project.

which main? the main of the example file?
I try to select it, but it cannot be selected as root function.
 
What can I do ?

xkardec, 2018-10-17:
[image]

@VishalX if I click on any of those `main`, nothing happens.
What can I do to solve this?
I've tried on both Windows and Ubuntu.

salcanmor, 2018-10-17:
I'm experiencing that same error.

VishalX, 2018-10-19:
@xkardec 

If you're following the [BUILD_USING_SDX_GUI.md](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md), you don't have to select this. Just make sure you don't have any other `main` in your project.

Also, `main1` is unnecessary and will be removed from the sources.

xkardec, 2018-10-19:
@VishalX 

I'm experiencing the same error than https://github.com/Xilinx/CHaiDNN/issues/60#issuecomment-429765252 

If I do not specify which the root function is, what I got is: 
`ERROR: [SYNCHK 200-79] Cannot find the top function '' in the design. Possible causes are: (1) the top function name is misspelled; (2) the top function is nonexistent or declared as static.`

How can I solve this problem?

VishalX, 2018-10-19:
@xkardec 

The `top` function and the `root` functions are different. 
The error suggests that you're not specifying the `Top` function.  Please check the build settings ( step 16 ) in [BUILD_USING_SDX_GUI.md](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md). Provide the correct absolute path to `xi_convolution_top.cpp` file.

xkardec, 2018-10-19:
> @xkardec
>
> The `top` function and the `root` functions are different.
> The error suggests that you're not specifying the `Top` function. Please check the build settings ( step 16 ) in [...]

@VishalX how can I specify the top function? I have followed step 16.

VishalX, 2018-10-22:
@xkardec,

Step 16 will automatically move `XiConvolutionTop` to hardware. To move other functions to hardware, please follow step 26.

VishalX, 2018-10-29:
Closing. Please re-open if required.

closed 2018-10-29 by VishalX