Xilinx/CHaiDNN #78: root (defualt)ZCU102 platform is not seen in the selection

adu81020799, 2018-10-18:
Hello,
I was trying to generate a custom hardware using the guide here
https://github.com/Xilinx/CHaiDNN/blob/master/docs/CUSTOM_PLATFORM_GEN.md
After I could generate the custom hardware and follow the step the custom platform did get added to the platform selection list but then the old ZCU102 platform is no more seen the selection guide. But in the installation directory-sdx /platforms shows the files for ZCU102. 
How can i get the old the root the platform of ZCU102.
Regards
Adarsh Sawant

VishalX, 2018-10-19:
@adu81020799 

Which version of SDx you're using?

adu81020799, 2018-10-19:
sdx 2018.2

VishalX, 2018-10-23:
@adu81020799,

Did you try the other options on the platform selection dialog?
Try clicking on `Manage Platform Repositories` and select `SDSoc` tab. 

[image]

VishalX, 2018-11-02:
Closing. Please reopen if required.

closed 2018-11-02 by VishalX