Xilinx/CHaiDNN #84: vgg example exits with segmentation violation

rbenshushan, 2018-10-22:
Hi,
I'd compiled vgg.cpp example using SDx GUI,  but after few iterations one thread exits with Signal 11; Signal: Segmentation violation.
Steps to reproduce:
1. Compile vgg_ex.cpp example in Debug mode.
2. Start Debug session.
3. After few iterations one thread return with segmentation violation.
Debug cursor pointed to line 632 (f_val = caffe_buff[off]) in xi_format_converter_top.cpp.
Resuming session again causes termination with exit code 255.

Moreover, I also compiled the vgg_ex.cpp in Release mode, It exits with segmentation fault.
Steps to reproduce:
1. Compile vgg_ex.cpp example in Release mode.
2. Copy the outputs in SD_CARD.
3. Reboot the system.
4. Launch the *.elf with an appropriate parameters
   ( e.x : <exefilename>.elf  Xilinx 6 <path to first image> <path to second image>)
5. After few iterations it exits with segmentation fault. 

I attached hereby output printings of both Debug and Release modes  

Waiting for response,
Thanks in advance,
Roee.
[image]
[image]

AnushaPerla, 2018-10-23:
Hi @rbenshushan

Can you share more details on design configuration and hardware you are using. Also share details on model and proto files.

We recommend you to try pre-built binaries with example models that are shared in modelzoo.

AnushaPerla, 2018-11-02:
Closing, Please re-open if required.

closed 2018-11-02 by AnushaPerla