Xilinx/CHaiDNN #86: Problem of executing XportDNN.pyc?

lishen565, 2018-10-23:
Hi all, recently I tried to quantilize my network by excecuting XportDNN.pyc as the https://github.com/Xilinx/CHaiDNN/blob/master/docs/QUANTIZATION.md and the  SETUP_TOOLS.md.
Something was going wrong after typing $ python XportDNN.pyc --quant_type "Xilinx" \
--deploy_model ./models/bvlc_googlenet_without_lrn/bvlc_googlenet_without_lrn_deploy.prototxt \
--weights ./models/bvlc_googlenet_without_lrn/bvlc_googlenet_without_lrn.caffemodel \
--quantized_deploy_model ./models/bvlc_googlenet_without_lrn/RistrettoDemo/bvlc_googlenet_without_lrn_quantized_deploy.prototxt \
--calibration_directory ./data/ilsvrc12/ILSVRC2012_img_val --calibration_size 32 \
--bitwidths 6,6,6 --dims 3,224,224 --transpose 2,0,1 \
--channel_swap 2,1,0 --raw_scale 255.0 \
--mean_value 104,117,123 --input_scale 1.0,
which said "ImportError: libprotobuf.so.9: cannot open shared object file: No such file or directory".
Did anybody meet this situation before and how to solve this issue?Thanks a lot.
 The detail info is below:
 
[image]

smenon009, 2018-10-24:
Hi @lishen565 

This looks like a python import error. 
Were you able to build caffe successfully?($make all -j 8 && make distribute)

zjchenchujie, 2018-11-20:
please setup environment first by executing:

#!/bin/bash

# Install the general dependencies
sudo apt-get update 
sudo apt-get -y upgrade
sudo apt-get install -y python-pip
sudo apt install -y virtualenv

#Activate CHai virtualenv
virtualenv chaidnn_tools_ENV
source ./chaidnn_tools_ENV/bin/activate

# Install binary specific dependencies
sudo apt-get install -y libprotobuf-dev libleveldb-dev libsnappy-dev libopencv-dev libhdf5-serial-dev protobuf-compiler
sudo apt-get install -y --no-install-recommends libboost-all-dev
sudo apt-get install -y libatlas-base-dev
sudo apt-get install -y libgflags-dev libgoogle-glog-dev liblmdb-dev 
for req in $(cat ./tools_binaries/python/requirements.txt); do pip install $req; done 

source ./chaidnn_tools_ENV/bin/activate &

closed 2018-12-06 by VishalX