Xilinx/CHaiDNN #87: Performance and utilization measurement?

adu81020799, 2018-10-23:
Is there any document which shows how we can check the performance of Chai DNN in terms of latency and the power utilization and also utilization ?

VishalX, 2018-10-29:
@adu81020799,

The resource utilization reports are generated by SDx and can be found in the `Release/_sds` folder. 
To measure the performance, you can use the `sds_clock_counter()` function. Please refer to any of the example files for usage of `sds_clock_counter()` built-in function.

For details on measuring power on a particular device, I would recommend you to post your queries on [Xilinx SDSoC forum](https://forums.xilinx.com/t5/SDSoC-Environment-and-reVISION/bd-p/sdsoc).

adu81020799, 2018-10-30:
Hello, 
There are performance reports in the sds folder but there are no power reports in them. Is there any way to create power reports from the sdx tool as we have in the Vivado

VishalX, 2018-11-02:
@adu81020799 

Power reports are generated by the SDx tool (inside `_sds/p0/vivado/prj/prj.runs/impl_1`) but for accurate power measurements, running the design on the board would be best.

Also, you can open the project on Vivado by opening the project file `_sds/p0/vivado/prj/prj.xpr`.

averr5, 2019-05-01:
@adu81020799 @VishalX I am running Alexnet on the board and got the performance in images/s. 

Is there a way to measure the SW cycles (on PS alone) and the SW + HW accelerated cycles (to know how much it gets better by running it on the PL) like we get in SD Estimate?