Xilinx/CHaiDNN #89: Build CHaiDNN software failure for ZCU106 platform.

qsqqsqqsq, 2018-10-25:
Hi. I try to build CHaiDNN software for ZCU106. I followed the tips on https://github.com/Xilinx/CHaiDNN page. First I built the hardware,  I updated two files, Makefile and xi_conv_config.h to meet the requiement for ZCU106. I built the hardware successfully and got the BOOT.BIN and libxlnxdnn.so.
But when i build the software I got the following error:

## Compiling AlexNet

aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/home/albert/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/home/albert/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/alexnet_ex.cpp -o alexnet.elf
/home/albert/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/home/albert/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../SD_Card/lib/../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
../../SD_Card/lib/libxstack.so: undefined reference to `DeconvWrapper(short*, short*, short*, unsigned long long*, int*)'
../../SD_Card/lib/libxstack.so: undefined reference to `PoolWrapper(ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, int*)'
collect2: error: ld returned 1 exit status
example.mk:26: recipe for target 'alexnetfcn.elf' failed
make[1]: *** [alexnetfcn.elf] Error 1
make[1]: *** Waiting for unfinished jobs....
//homehome//albertalbert//XilinxXilinx//SDKSDK//2018.22018.2//gnugnu//aarch64aarch64//lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-lin/gnuaarch64-/7.2.1/..linux/bin/./../.../../aarch64-linux-gnu/bin/ld: /lib/gcc/skipping incompatible ..aarch64/../lib-/libpthread.so.0 whenlinux- gnu/7.2.1/../../../../aarch64-linux-searching gnu/bin/ld: for .skipping./ incompatible.. ./lib.//libpthread.so.0../lib
//libpthread.so.0home when/albert searching for ..//../Xilinx/SDK/lib/libpthread.so.02018.2/gnu
/home/aarch64//albert/lin/aarch64Xilinx/SDK/-linux/2018.2/gnu/bin/..aarch64/lin//lib/aarch64-linux/gcc/aarch64bin-/.linux-./lib/gnugcc//7.2.1aarch64-linux-gnu/7.2.1/./.././..././....//....//aarch64-aarch64-linux-linux-gnugnu/bin//bin/ld: ld: skippingskipping incompatible  ..incompatible /.../SD_Card.//lib/....//../lib/libpthread.so.0 SD_Cardwhen /libsearching for ..//../lib/..libpthread.so.0
./..././lib/./SD_Cardlibpthread.so.0 when/lib/ searching libxstack.so: for .undefined reference./.. to /lib/`DeconvWrapper(libpthread.so.0
short*../, short../*, shortSD_Card/lib*, unsigned/libxstack.so: long  undefined long*,reference to  int*`DeconvWrapper()'
.short*, .//home./.albert//SD_CardshortXilinx*/,SDK/ /short2018.2*/,gnu /unsignedaarch64 /longlin /longlibaarch64*/,libxstack.so : intundefined*)- linuxreference/'bin 
to. .`/PoolWrapper.(/.ap_uint/<SD_Card..128/>lib/*/libgcc/,libxstack.so ap_uint:< 128>undefined/ *reference, aarch64to  ap_uint`<PoolWrapper128(>-linuxap_uint-<gnu128/>7.2.1*/,. .ap_uint/<.128.>/*.,. /ap_uint.<.128/>aarch64*-,linux -ap_uintgnu</128bin>/*ld,:  ap_uintskipping< 128incompatible> *.,. /int.*.)/'lib
/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/home/albert/Xilinx/SDK/2018.2/gnu/collect2: error: ld returned 1 exit status
aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../SD_Card/lib/..*/.,. /ap_uintlib</128libpthread.so.0> *when,  searchingap_uint <for128 >.*.,/ .int.*/)lib'/
libpthread.so.0
../../SD_Card/lib/libxstack.so: undefined reference to `DeconvWrappercollect2: error: ld returned 1 exit status
(short*, short*, short*, unsigned long long*, int*)'
../../SD_Card/lib/libxstack.so: undefined reference to `PoolWrapper(ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, ap_uint<128>*, int*)'
collect2: error: ld returned 1 exit status
example.mk:21: recipe for target 'googlenet.elf' failed
make[1]: *** [googlenet.elf] Error 1
example.mk:31: recipe for target 'alexnet.elf' failed
make[1]: *** [alexnet.elf] Error 1
example.mk:46: recipe for target 'resnet50.elf' failed
make[1]: *** [resnet50.elf] Error 1
make[1]: Leaving directory '/home/albert/CHaiDNN/software/scripts'
Makefile:46: recipe for target 'all' failed
make: *** [all] Error 2

VIVADO version is 2018.2，CHaiDNN is the latest version. I did not understand why there is no define for PoolWrapper and DeconvWrapper. 

It would be really nice if you can help me resolve this issue.

Thanks,
Albert.

AnushaPerla, 2018-10-25:
Hi @qsqqsqqsq 

Please ensure that the software and the hardware are build with the same settings. 

It seems while building hardware, you have disabled pool and deconv kernels. So try to build software also by disabling pool and deconv. To exclude pool and deconv Kernel, set POOL_ENABLE=0 & DECONV_ENABLE=0 in Makefile.

For more details, please see the [Run Inference](https://github.com/Xilinx/CHaiDNN#run-inference)

qsqqsqqsq, 2018-10-26:
Hi @AnushaPerla
I've checked the hardware Makefile:
# Main Variables
LIBRARY     := libxlnxdnn.so
PLATFORM    := zcu106
CLOCK_ID    := 1 # Core clock ID
DM_CLOCK_ID := 1 # Data Motion Clock ID
# Below path should be relative
PROJ_PATH   := ..
# H/W Kernel enables
DIET_CHAI_Z	     ?= 0
DIET_CHAI_ZUPLUS ?= 0

CONV_ENABLE   ?= 1
POOL_ENABLE   ?= 1
DECONV_ENABLE ?= 1

So I've set the POOL_ENABLE=1 & DECONV_ENABLE=1. I also tried to set the POOL_ENABLE=0 & DECONV_ENABLE=0 in Software Makefile. There comes a different error:

## Compiling AlexNet

aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=0  -D__DECONV_ENABLE__=0 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/home/albert/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/home/albert/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/alexnet_ex.cpp -o alexnet.elf
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../SD_Card/lib/../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find ../../lib/libpthread.so.0
collect2: error: ld returned 1 exit status
example.mk:26: recipe for target 'alexnetfcn.elf' failed
make[1]: *** [alexnetfcn.elf] Error 1
make[1]: *** Waiting for unfinished jobs....
//usrusr//liblib//gccgcc--crosscross//aarch64aarch64--linuxlinux--gnugnu//55//....//....//....//....//aarch64aarch64--linuxlinux--gnugnu//binbin//ldld::  skippingskipping  incompatibleincompatible  ....//....//liblib//libpthread.so.0libpthread.so.0  whenwhen  searchingsearching  forfor  ....//....//liblib//libpthread.so.0libpthread.so.0

//usrusr//liblib//gccgcc--crosscross//aarch64aarch64--linuxlinux--gnugnu//55//....//....//....//....//aarch64aarch64--linuxlinux--gnugnu//binbin//ldld::  skippingskipping  incompatibleincompatible  ....//....//SD_CardSD_Card//liblib//....//....//liblib//libpthread.so.0libpthread.so.0  whenwhen  searchingsearching  forfor  ....//....//liblib//libpthread.so.0libpthread.so.0

//usrusr//liblib//gccgcc--crosscross//aarch64aarch64--linuxlinux--gnugnu//55//....//....//....//....//aarch64aarch64--linuxlinux--gnugnu//binbin//ldld::  cannotcannot  findfind  ....//....//liblib//libpthread.so.0libpthread.so.0

collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
example.mk:46: recipe for target 'resnet50.elf' failed
make[1]: *** [resnet50.elf] Error 1
example.mk:31: recipe for target 'alexnet.elf' failed
make[1]: *** [alexnet.elf] Error 1
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible ../../SD_Card/lib/../../lib/libpthread.so.0 when searching for ../../lib/libpthread.so.0
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find ../../lib/libpthread.so.0
collect2: error: ld returned 1 exit status
example.mk:21: recipe for target 'googlenet.elf' failed
make[1]: *** [googlenet.elf] Error 1
make[1]: Leaving directory '/home/albert/CHaiDNN/software/scripts'
Makefile:46: recipe for target 'all' failed
make: *** [all] Error 2

So there must be more than one error in my project.

VishalX, 2018-10-29:
@qsqqsqqsq,

Are you using the compiler & libraries from SDx tool? 
It seems there is a problem with the libpthread.so (as the error says: `skipping incompatible`). 
Could you please re-check whether the compiler is using the correct paths?

VishalX, 2019-01-09:
Closing. Please reopen if required.

closed 2019-01-09 by VishalX