Xilinx/CHaiDNN #90: Running tiny darknet with chai.

adu81020799, 2018-10-25:
Hello, 
I would like to build tiny darknet , but the prototxt file  given from tiny darknet is in different format(from what Chai DNN expects). I wanted to know is there a way I can convert dark net format  into caffe format(which chai expects). I am trying to do this as tiny darknet is really small network and if we could make this we can use on smaller platforms like zynq.Plus i have seen the reports generated for the yolo for Chai DNN. So there must be a way. 
Regards
Adarsh

adu81020799, 2018-10-26:
Hi Guys I used the following converter to convert darknet framework to caffe model.
follow the link till two steps
https://github.com/ysh329/darknet2caffe.

salcanmor, 2018-10-30:
did you make yolo working in CHaiDNN ?
If yes, which version of Yolo ?

adu81020799, 2018-10-30:
No not yet. Iam still struggling with tiny darknet.Did you ?

adu81020799, 2018-10-30:
Hello,
Today I could compile my prototxt file for Darknet with Chai DNN quantizer.The only concern I have rite now is the warning that gets generated for some of the layers
20 80
layer7_bn, BatchNorm
['layer7_conv'], ['layer7_conv']
bw_layer_in:  6
th_layer_in:  1794.7390563488007
bw_layer_out:  6
th_layer_out:  1257.0803022682667

WARNING: Previous layer's th_out is not equal to current layer's layer7_bnth_in.
WARNING: A drop in accuracy may occur.

Can some one let me know what is the warning about? And is there is possibility to get around this warning ?

adu81020799, 2018-11-04:
Further I am having problem with the make file:
Here is the make file I used:
ARM_CXX = aarch64-linux-gnu-g++    
#SDx_BUILD_PATH = /opt/Xilinx/SDx/2018.2
SDx_BUILD_PATH =/proj/xbuilds/2018.2_daily_latest/installs/lin64/SDx/2018.2
IDIRS = -I$(SDx_BUILD_PATH)/target/aarch64-linux/include -I/home/uncc_embed/CHaiDNN/SD_Card/opencv/arm64/include
PB_ARM_DIR = SD_Card/protobuf_arm64
OPENCV_DIR = SD_Card/opencv_arm64
CBLAS_ARM_DIR = SD_Card/cblas_arm64
OPENCV_LIBS = -lopencv_core -llzma -ltiff -lpng16 -lz -ljpeg -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -lwebp
LDIRS = -L/home/uncc_embed/CHaiDNN/SD_Card/opencv/arm64/lib -L/home/uncc_embed/CHaiDNN/SD_Card/protobuf/arm64/lib -L/home/uncc_embed/CHaiDNN/SD_Card/cblas/arm64/lib
LLIBS = -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm
CFLAGS_ARM = -std=c++11 -D__SDSOC=1 -Wno-write-strings
.PHONY: all
MyNet.elf : ./MyNet.cpp
	$(ARM_CXX) $(CFLAGS_ARM) -L$(PB_ARM_DIR)/lib -I$(PB_ARM_DIR)/include -L$(OPENCV_DIR)/lib -I$(OPENCV_DIR)/include -L$(CBLAS_ARM_DIR)/lib -I$(CBLAS_ARM_DIR)/include $(IDIRS) $(LDIRS) $(LLIBS) $(OPENCV_LIBS) $^ -o $@

The error I get is:
 make MyNet.elfaarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -LSD_Card/protobuf_arm64/lib -ISD_Card/protobuf_arm64/include -LSD_Card/opencv_arm64/lib -ISD_Card/opencv_arm64/include -LSD_Card/cblas_arm64/lib -ISD_Card/cblas_arm64/include -I/proj/xbuilds/2018.2_daily_latest/installs/lin64/SDx/2018.2/target/aarch64-linux/include -I/home/uncc_embed/CHaiDNN/SD_Card/opencv/arm64/include -LSD_Card/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopencv_core -llzma -ltiff -lpng16 -lz -ljpeg -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -lwebp MyNet.cpp -o MyNet.elf
MyNet.cpp:16:10: fatal error: sds_lib.h: No such file or directory
 #include "sds_lib.h"
          ^~~~~~~~~~~
compilation terminated.
Makefile:14: recipe for target 'MyNet.elf' failed
make: *** [MyNet.elf] Error 1
Please let me know what are the problems witth the make file.

VishalX, 2019-01-09:
Please reopen if required.

closed 2019-01-09 by VishalX

averr5, 2019-02-20:
@adu81020799 How did you get through - fatal error: sds_lib.h: No such file or directory ?
I am facing this error while i am trying to build the software with the Makefile